RoboStack/ros-galactic #5: Set environment variables for Galactic

Tobias-Fischer, 2021-07-10:


traversaro, 2021-07-18:
Indeed, `ros2 doctor` fails with some errors related to this: 
~~~
(ros_galactic) straversaro@IITICUBLAP102:~$ ros2 doctor
/home/straversaro/mambaforge/envs/ros_galactic/lib/python3.8/site-packages/ros2doctor/api/network.py: 71: UserWarning: ERROR: No loopback IP address is found.
/home/straversaro/mambaforge/envs/ros_galactic/lib/python3.8/site-packages/ros2doctor/api/network.py: 74: UserWarning: Only loopback IP address is found.
/home/straversaro/mambaforge/envs/ros_galactic/lib/python3.8/site-packages/ros2doctor/api/network.py: 77: UserWarning: No multicast IP address is found.
/home/straversaro/mambaforge/envs/ros_galactic/lib/python3.8/site-packages/ros2doctor/api/package.py: 40: UserWarning: ERROR: ROS_DISTRO is not set.
/home/straversaro/mambaforge/envs/ros_galactic/lib/python3.8/site-packages/ros2doctor/api/package.py: 150: UserWarning: ERROR: distro packages info is not found.
/home/straversaro/mambaforge/envs/ros_galactic/lib/python3.8/site-packages/ros2doctor/api/__init__.py: 112: UserWarning: Fail to call PackageCheck class functions.
/home/straversaro/mambaforge/envs/ros_galactic/lib/python3.8/site-packages/ros2doctor/api/platform.py: 37: UserWarning: ERROR: ROS_DISTRO is not set.
/home/straversaro/mambaforge/envs/ros_galactic/lib/python3.8/site-packages/ros2doctor/api/platform.py: 69: UserWarning: ERROR: Missing rosdistro info. Unable to check platform.

2/3 check(s) failed

Failed modules: network platform
~~~

traversaro, 2021-07-18:
https://github.com/RoboStack/ros-galactic/issues/14 suggests that also `AMENT_PREFIX_PATH`  should be set.

traversaro, 2021-07-19:
As a workaround for the moment, it is possible to set manually the env variables, for example in Linux:
~~~
source ${CONDA_PREFIX}/setup.sh
~~~

After doing so, the output of `ros2 doctor` is: 
~~~
/home/straversaro/mambaforge/envs/ros2/lib/python3.8/site-packages/ros2doctor/api/network.py: 71: UserWarning: ERROR: No loopback IP address is found.
/home/straversaro/mambaforge/envs/ros2/lib/python3.8/site-packages/ros2doctor/api/network.py: 74: UserWarning: Only loopback IP address is found.
/home/straversaro/mambaforge/envs/ros2/lib/python3.8/site-packages/ros2doctor/api/network.py: 77: UserWarning: No multicast IP address is found.
/home/straversaro/mambaforge/envs/ros2/lib/python3.8/site-packages/ros2doctor/api/package.py: 112: UserWarning: ament_index_cpp has been updated to a new version. local: 1.0.6 < latest: 1.2.0
/home/straversaro/mambaforge/envs/ros2/lib/python3.8/site-packages/ros2doctor/api/package.py: 112: UserWarning: ament_index_python has been updated to a new version. local: 1.0.6 < latest: 1.2.0
~~~

closed 2021-07-22 by Tobias-Fischer