RoboStack/ros-galactic #13: Error following README instructions in rosdep init

traversaro, 2021-07-18:
I followed the instructon in the README, and at the `rosdep init`  I get an error:
~~~
(ros_galactic) straversaro@IITICUBLAP102:~$ rosdep init
ERROR: cannot create /etc/ros/rosdep/sources.list.d:
        [Errno 13] Permission denied: '/etc/ros'
~~~

<br/>
Environment (<code>conda list</code>):
<details>

```
$ conda list
# packages in environment at /home/straversaro/mambaforge/envs/ros_galactic:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       1_gnu    conda-forge
alsa-lib                  1.2.3                h516909a_0    conda-forge
argcomplete               1.12.3             pyhd8ed1ab_2    conda-forge
assimp                    5.0.1                hedfc422_5    conda-forge
atk-1.0                   2.36.0               h3371d22_4    conda-forge
attr                      2.4.48               h516909a_0    conda-forge
attrs                     21.2.0             pyhd8ed1ab_0    conda-forge
binutils                  2.36.1               hdd6e379_1    conda-forge
binutils_impl_linux-64    2.36.1               h193b22a_1    conda-forge
binutils_linux-64         2.36                hf3e587d_32    conda-forge
boost                     1.74.0           py38hc10631b_3    conda-forge
boost-cpp                 1.74.0               h312852a_4    conda-forge
bullet                    3.17                 ha770c72_0    conda-forge
bullet-cpp                3.17                 h1abd341_0    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.17.1               h7f98852_1    conda-forge
c-compiler                1.2.0                h7f98852_0    conda-forge
ca-certificates           2021.5.30            ha878542_0    conda-forge
cairo                     1.16.0            h6cf1ce9_1008    conda-forge
catkin_pkg                0.4.23             pyh9f0ad1d_0    conda-forge
catkin_tools              0.6.1            py38h578d9bd_3    conda-forge
certifi                   2021.5.30        py38h578d9bd_0    conda-forge
cffi                      1.14.6           py38ha65f79e_0    conda-forge
cmake                     3.21.0               h8897547_0    conda-forge
compilers                 1.2.0                ha770c72_0    conda-forge
console_bridge            1.0.1                h4bd325d_0    conda-forge
cppcheck                  2.5              py38hbffb2f6_0    conda-forge
cryptography              3.4.7            py38ha5dfef3_0    conda-forge
cxx-compiler              1.2.0                h4bd325d_0    conda-forge
cycler                    0.10.0                     py_2    conda-forge
dbus                      1.13.6               h48d8840_2    conda-forge
distro                    1.5.0              pyh9f0ad1d_0    conda-forge
docutils                  0.17.1           py38h578d9bd_0    conda-forge
eigen                     3.3.9                h4bd325d_1    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.1                h9c3ff4c_0    conda-forge
ffmpeg                    4.3.1                hca11adc_2    conda-forge
flake8                    3.9.2              pyhd8ed1ab_0    conda-forge
flann                     1.9.1             h2e58136_1008    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            hba837de_1005    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
foonathan-memory          0.6.2                he1b5a44_1    conda-forge
fortran-compiler          1.2.0                h1990efc_0    conda-forge
freeimage                 3.18.0               h88c329d_7    conda-forge
freetype                  2.10.4               h0708190_1    conda-forge
fribidi                   1.0.10               h36c2ea0_0    conda-forge
gcc_impl_linux-64         9.3.0               h70c0ae5_19    conda-forge
gcc_linux-64              9.3.0               hf25ea35_32    conda-forge
[... 491 lines cut ...]
          conda version : 4.10.3
    conda-build version : not installed
         python version : 3.9.5.final.0
       virtual packages : __linux=4.19.104=0
                          __glibc=2.31=0
                          __unix=0=0
                          __archspec=1=x86_64
       base environment : /home/straversaro/mambaforge  (writable)
      conda av data dir : /home/straversaro/mambaforge/etc/conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/robostack-experimental/linux-64
                          https://conda.anaconda.org/robostack-experimental/noarch
                          https://conda.anaconda.org/conda-forge/linux-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : /home/straversaro/mambaforge/pkgs
                          /home/straversaro/.conda/pkgs
       envs directories : /home/straversaro/mambaforge/envs
                          /home/straversaro/.conda/envs
               platform : linux-64
             user-agent : conda/4.10.3 requests/2.25.1 CPython/3.9.5 Linux/4.19.104-microsoft-standard ubuntu/20.04.2 glibc/2.31
                UID:GID : 1000:1000
             netrc file : None
           offline mode : False
```
</details>

traversaro, 2021-07-18:
Related to https://github.com/RoboStack/ros-galactic/issues/7 .

fyi @Tobias-Fischer @gregfreeman

traversaro, 2021-07-18:
Possibly a duplicate of https://github.com/RoboStack/ros-galactic/issues/1 .

traversaro, 2021-07-19:
I just tried after manually sourcing the environment via the commands described in https://github.com/RoboStack/ros-galactic/issues/5#issuecomment-882720594 , but the issue persists. On Windows a file outside the environment is created, but it is not creating an error as apparently I have the permission to do so:
~~~
(ros2) C:\Users\STraversaro>rosdep init
Wrote /etc/ros\rosdep\sources.list.d\20-default.list
Recommended: please run

        rosdep update
~~~

wolfv, 2021-07-19:
but the rosdep is essentially the same as for noetic, isn't it?

nuclearsandwich, 2021-07-19:
> ERROR: cannot create /etc/ros/rosdep/sources.list.d:
>         [Errno 13] Permission denied: '/etc/ros'

This looks like rosdep init is failing to run because it is trying to write to the root `/etc/ros` rather than a conda environment. As far as I know there is no way to avoid this without the changes proposed in https://github.com/ros-infrastructure/rosdep/pull/583

How is that working for the noetic instructions?

traversaro, 2021-07-19:
I guess it is somehow related to this patch: https://github.com/RoboStack/ros-noetic/blob/cbd8a2011de90923a154bca7533d0e1578269065/patch/ros-noetic-rosdep.patch . However, I tought rosdep is not package by robostack but in conda-forge, let me check that.

traversaro, 2021-07-19:
Based on the rosdep patch in conda-forge, I guess we need to set `ROS_ETC_DIR` somehow, see https://github.com/conda-forge/rosdep-feedstock/blob/master/recipe/add-robostack.patch#L326 .

traversaro, 2021-07-19:
At a first glance, it seems that in ROS1 `ROS_ETC_DIR` is defined in https://github.com/ros/ros_environment/blob/noetic/env-hooks/1.ros_etc_dir.sh.em, that then is copied in the correct location by https://github.com/RoboStack/ros-noetic/blob/6aee2cd0311ebe70219cd32bf85e67a313bb7e0d/additional_recipes/build_catkin.sh#L85 . 

This is a related issue: https://github.com/ros/ros_environment/issues/25 .

traversaro, 2021-07-19:
Ok, it seems that the options are:
* either We define `ROS_ETC_DIR` in some package (as a local package in `ros-galactic-workspace`?)
* or We backport https://github.com/ros-infrastructure/rosdep/pull/583/files to rosdep in conda-forge, and we define  `ROSDEP_ETC_DIR` somewhere (see the previous point)

Do you see any other possible solution @Tobias-Fischer @nuclearsandwich ?

Tobias-Fischer, 2021-07-19:
It's already patched in the conda-forge feedstock: https://github.com/conda-forge/rosdep-feedstock/blob/master/recipe/add-robostack.patch

So we just need to define `ROSDEP_ETC_DIR` somewhere, similarly to https://github.com/ros/ros_environment/blob/noetic/env-hooks/1.ros_etc_dir.sh.em

Tobias-Fischer, 2021-07-20:
Fix suggested in https://github.com/RoboStack/vinca/pull/13

Tobias-Fischer, 2021-07-20:
After rebuilding the workspace package that should be fixed

Tobias-Fischer, 2021-07-22:
Works fine

closed 2021-07-22 by Tobias-Fischer