RoboStack/ros-galactic #21: Add a mutex package with some global pins

wolfv, 2021-07-19:


Tobias-Fischer, 2021-07-22:
Can we get rid of version specifiers in the robostack.yaml using the mutex package? Then we can have a combined robostack.yaml for noetic and galactic.

Currently the robostack.yaml's differ for:
- ogre (no specifier on galactic, 1.10 in noetic)
- pcl  (no specifier on galactic, 1.11 in noetic)
- pyqt (5 on galactic, 5.12 in noetic)

Having a single robostack.yaml seems important, as we use a single rosdep on conda and that cannot distinguish robostack-noetic from robostack-galactic easily as far as I can tell.

wolfv, 2021-07-23:
yeah that makes a lot of sense. We should have pins in the `conda_build_config` and in the mutex package only.

Tobias-Fischer, 2021-08-13:
See #48