RoboStack/ros-galactic #23: Fix numpy include in rosidl_generator_py

Tobias-Fischer, 2021-07-19:
Currently finds the system numpy

Tobias-Fischer, 2021-07-20:
See https://github.com/Tobias-Fischer/rosidl_python/commit/1c353799377ae5e294226ef67ad877777bc4a9d6 for a potential solution. What do you think @wolfv @traversaro?

wolfv, 2021-07-20:
I think @tfoote was also looking into this yesterday. Maybe he's got an opinion

traversaro, 2021-07-20:
> 
> 
> See [Tobias-Fischer/rosidl_python@1c35379](https://github.com/Tobias-Fischer/rosidl_python/commit/1c353799377ae5e294226ef67ad877777bc4a9d6) for a potential solution. What do you think @wolfv @traversaro?

`FindPythonInterp` is deprecated at the CMake level, and `FindPython3` is instead the recommended way to go, see https://cmake.org/cmake/help/v3.21/module/FindPythonInterp.html , so I fully agree!

Tobias-Fischer, 2021-07-20:
I think the more critical fix is that currently the numpy path is only added on OSX and Win .. the change to FindPython3 is cosmetics on top ;)

Tobias-Fischer, 2021-07-20:
Should have mentioned this earlier

traversaro, 2021-07-20:
Ahh, that was a tricky one. In any case it make sense.

tfoote, 2021-07-20:
I believe that this is the right fix. The additional path is the critical part, but the new API is much cleaner.

Jumping to the Python3 find logic is much cleaner. I'm confirming with @cottsay whether or not the minimum of 3.14 is ok to use(That's when the new api was added https://cmake.org/cmake/help/latest/module/FindPython3.html#result-variables )

As of Galactic only RHEL is a supported platform that doesn't have a new enough CMake Version:
https://www.ros.org/reps/rep-2000.html#id38 to jump forward. And they may have a backports available or we can move forward.

tfoote, 2021-07-20:
It looks like RHEL 8.4 is using CMake 3.18 so we should be able to push forward. @cottsay will be working to update the REP.

closed 2021-07-20 by Tobias-Fischer

Tobias-Fischer, 2021-07-20:
This seems to be working nicely now: 
`-- Using numpy include directory: $PREFIX/lib/python3.8/site-packages/numpy/core/include`

PR #30 (2021-07-20): Fix rosidl-generator-py numpy include 
Fix #23