RoboStack/ros-galactic #29: ros2 docker, packages that use rclpy aren't working on MacOS

raghavauppuluri13, 2021-07-20:
https://github.com/ros2/rclpy/issues/805

Tobias-Fischer, 2021-07-20:
@raghavauppuluri13 - can you please try the latest updated build of rclpy? It should work.

raghavauppuluri13, 2021-07-20:
Error still there unfortunately. I ran `mamba update --all`, reset env, and ran `ros2 doctor` and error is still there. Anything to do with pybind11?

Tobias-Fischer, 2021-07-20:
Ah, annoying. I'll have a look later.

Tobias-Fischer, 2021-07-21:
Can you try `import rclpy` in a python session? This is working fine for me on osx. For `ros2 doctor`, I'm running into #39 ..

Tobias-Fischer, 2021-07-21:
Indeed I can now replicate the issue on `ros2 doctor`

Tobias-Fischer, 2021-07-21:
I think this change requires a rebuild of all packages that depend on rosidl-generator-py ... https://github.com/RoboStack/ros-galactic/commit/31e0d4866183cbc558af28adb4642480f1639d66

Tobias-Fischer, 2021-07-21:
Another good use case of the prototype that you have written @wolfv

Tobias-Fischer, 2021-07-22:
See https://github.com/RoboStack/ros-galactic/issues/39 instead

closed 2021-07-22 by Tobias-Fischer