RoboStack/ros-galactic #31: Upstream patches

Tobias-Fischer, 2021-07-20:
We should try and get as many patches upstream as possible.

traversaro, 2022-01-21:
Once merged and released https://github.com/ros2/rviz/pull/817 should permit to remove the patch https://github.com/RoboStack/ros-galactic/blob/main/patch/ros-galactic-rviz-assimp-vendor.patch

traversaro, 2022-02-06:
`ros-galactic-osrf-testing-tools-cpp.patch` proposed upstream in https://github.com/osrf/osrf_testing_tools_cpp/pull/68 .

traversaro, 2022-02-06:
> Once merged and released [ros2/rviz#817](https://github.com/ros2/rviz/pull/817) should permit to remove the patch https://github.com/RoboStack/ros-galactic/blob/main/patch/ros-galactic-rviz-assimp-vendor.patch

Actually we probably need to wait for a galactic backport if anyone ever works on that: https://github.com/ros2/rviz/pull/826 .

closed 2025-02-10 by traversaro