RoboStack/ros-galactic #44: The road to all packages on Linux

Tobias-Fischer, 2021-07-28:
The following dependencies are missing on conda-forge:
- 'coinor-libclp-dev'
- 'coinor-libcgl-dev
- 'coinor-libcoinutils-dev'
- 'libgps'
- 'libmongoclient-dev'
- 'spirv-headers'
- 'spirv-tools'
- 'libbluetooth'
- 'libbluetooth-dev'
- 'libvulkan-dev'
- 'v4l-utils'
- 'cwiid'
- 'spacenavd'
- 'libqglviewer2-qt5'
- 'libqglviewer-dev-qt5'
- 'cwiid-dev'

There is something weird going on with the `swri*` packages for which recipes currently cannot be created. I added this hack in vinca:
https://github.com/RoboStack/vinca/blob/54870d5947e30c0a6054d1452a88b3f9710dc960/vinca/main.py#L438-L439

Packages that do not yet work:
 - cartographer (needs updating, cannot find lua)
 - cartographer-ros (needs cartographer)
 - embree-vendor (needs fixing of cmake scripts, cannot find tbb)
 - fmilibrary-vendor fails with `multiple rules generate ExpatEx/libexpat.a`
 - fmi-adapter needs `fmilibrary-vendor`
 - fmi-adapter-examples needs fmi-adapter
 - gpsd-client needs `libgps`
 - gps-umd needs gpsd-client
 - mrpt2 needs cmake fixes, see https://github.com/RoboStack/ros-noetic/blob/master/patch/ros-noetic-mrpt2.patch
 - octovis needs `libQGLViewer`
 - popf needs FlexLexer.h
 - plansys2-popf-plan-solver needs popf
 - plansys2-* needs plansys2-popf-plan-solver
 - rc-dynamics-api refers to https://github.com/ros2-gbp/rc_dynamics_msgs/ which is not a repo anymore ..
 - rc-genicam-driver needs RC_GENICAM_API
 - ros2-socketcan fails with `error: 'sa_family_t' does not name a type`
 - rosbag2-bag-v2-plugins needs ros1_rosbag_storage (not sure where that comes from)
 - sdformat-urdf fails solving because of a missing migration of tinyxml2 (best guess)
 - spacenav needs `spnav`
 - tvm-vendor fails with `CMake Error: Unknown argument --configure`
 - ublox-dgnss-node fails with `cannot find -lusb-1.0`
 - ublox-dgnss needs ublox-dgnss-node
 - usb-cam fails with `cannot find -lavcodec` and `cannot find -lswscale`
 - warehouse-ros-mongo needs `MONGODB`
 - run-ompl-constrained-planning needs warehouse-ros-mongo 
 - wiimote needs `bluetooth`

/cc @wolfv

wolfv, 2021-07-29:
Cool, thanks for this writeup! 

The coin libraries should be available under 

coin-or-cgl etc on conda-forge (https://github.com/conda-forge/coin-or-cgl-feedstock)

I think spacenavd might also exist as it's used in FreeCAD.

And mongoclient could be part of mongo but no idea.

Looking forward to further hacking on this!

Tobias-Fischer, 2021-07-29:
Cool - check out https://github.com/RoboStack/ros-galactic/commit/68f7c3e184fd6d9cddfe673ec73108f81ee8895a

mongoclient is the old client for the mongo database, and not part of mongo AFAIR

traversaro, 2021-07-29:
> fmilibrary-vendor fails with `multiple rules generate ExpatEx/libexpat.a`

[fmi-library](https://github.com/modelon-community/fmi-library) is available in conda-forge as [fmilib](https://github.com/conda-forge/fmilib-feedstock), probably it is possible patch `fmilibrary-vendor` to look for it or something like that.

traversaro, 2021-07-29:
> [fmi-library](https://github.com/modelon-community/fmi-library) is available in conda-forge as [fmilib](https://github.com/conda-forge/fmilib-feedstock), probably it is possible patch [...]

I just checked that and the tricky aspect is that `fmi-library` does not install a `FMILibraryCMakeConfig.cmake`  or similar, so we would need to add to `fmilibrary-vendor` (I wrote one in https://github.com/robotology/gazebo-fmi/blob/master/cmake/FindFMILibrary.cmake).

Tobias-Fischer, 2021-07-29:
Yes, that's probably the way to go for that particular package.

Tobias-Fischer, 2021-07-29:
Thanks for the hint @traversaro: https://github.com/RoboStack/ros-galactic/commit/d064c7a5c064aa4279764d6edbf2ae991fdd1e78

wolfv, 2021-07-30:
I'm wondering if we should also add a pin to the conda build config for fmi library 🤔

Tobias-Fischer, 2021-07-31:
I got mrpt2 to work - what a pain: https://github.com/RoboStack/ros-galactic/commit/55e25f5d84326479aaa53725cb743ecbcf5312f9

Tobias-Fischer, 2021-07-31:
Re fmi: sounds like a good idea @wolfv

Tobias-Fischer, 2021-08-02:
For cartographer, we can do sth like this:
```diff
diff --git a/patch/dependencies.yaml b/patch/dependencies.yaml
index b0b0ee9..78593cb 100644
--- a/patch/dependencies.yaml
+++ b/patch/dependencies.yaml
@@ -1,3 +1,5 @@
+cartographer_ros:
+  add_host: ["cartographer"]
 ackermann_msgs:
   add_host: ["numpy"]
   add_run: ["numpy"]
diff --git a/vinca_linux_64.yaml b/vinca_linux_64.yaml
index 8a1e53a..4a0898a 100644
--- a/vinca_linux_64.yaml
+++ b/vinca_linux_64.yaml
@@ -13,7 +13,8 @@ packages_skip_by_deps:
   # - rviz
   # - diagnostic_updater
 
-packages_remove_from_deps: []
+packages_remove_from_deps:
+  - cartographer
   # - stage-ros
   # - python_qt_binding
   # - joint_state_publisher_gui
```

However, it needs https://github.com/conda-forge/cartographer-feedstock/pull/17

Tobias-Fischer, 2021-09-30:
See https://github.com/RoboStack/ros-galactic/issues/17

closed 2021-09-30 by Tobias-Fischer