RoboStack/ros-galactic #46: libBLAS.dylib contains the wrong version

Donglin-Wang2, 2021-07-29:
I have just followed the instruction in the README to set up the environment. However, it seem like every time I run ```ros2```  command or turtlesim, I run into the following error:
```
dyld: Library not loaded: /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/libBLAS.dylib
  Referenced from: /System/Library/Frameworks/Accelerate.framework/Versions/A/Frameworks/vecLib.framework/Versions/A/vecLib
  Reason: Incompatible library version: vecLib requires version 1.0.0 or later, but libBLAS.dylib provides version 0.0.0
```
Is there any way I can resolve this?

Tobias-Fischer, 2021-07-29:
Hi @Donglin-Wang2,

Many thanks for trying RoboStack :)! You are encountering the issue described in https://github.com/RoboStack/ros-galactic/issues/39 - we have a fix for it, but need to rebuild some more packages.

In the meantime, you can simply `unset DYLD_LIBRARY_PATH` after `conda activate robostackenv`. This should work - please let us know if not.

closed 2021-07-29 by Tobias-Fischer

Donglin-Wang2, 2021-07-29:
Hi Tobias,
Thanks for the timely response! I used the unset command after activating the environment. However, I ran into the following error when trying to run turtlesim:
```
dyld: lazy symbol binding failed: Symbol not found: __ZN6spdlog7details7log_msgC1ENS_10source_locEN3fmt2v717basic_string_viewIcEENS_5level10level_enumES6_
  Referenced from: /opt/miniconda3/envs/ros2/lib/librcl_logging_spdlog.dylib
  Expected in: /opt/miniconda3/envs/ros2/lib/libspdlog.1.dylib

dyld: Symbol not found: __ZN6spdlog7details7log_msgC1ENS_10source_locEN3fmt2v717basic_string_viewIcEENS_5level10level_enumES6_
  Referenced from: /opt/miniconda3/envs/ros2/lib/librcl_logging_spdlog.dylib
  Expected in: /opt/miniconda3/envs/ros2/lib/libspdlog.1.dylib
```

Tobias-Fischer, 2021-07-29:
Hi - sorry, this is another problem for which we have a fix but need a rebuild. You can temporarily fix it with `mamba install spdlog=1.8.5`

Donglin-Wang2, 2021-07-30:
Thank you so much for the response. I really appreciate the fact that you shared these script for ease of installation across all platform.