RoboStack/ros-galactic #50: ClobberError

PrasadNR, 2021-08-29:
What is this error and why do I see this?

[image]

I happen to see this for `mamba install ros-galactic-desktop` on Windows 8.

Tobias-Fischer, 2021-08-29:
It basically means that two packages (in the first example `ros-galactic-demo-nodes-py` and `ros-galactic-examples-rclpy-executors` both contain the same file `library/bin/talker.exe`. I am not sure whether the packages are still being installed? Could you please report the output of `mamba list`? If they are installed anyhow, I would suggest to ignore these errors; if you can't install RoboStack because of it we'll look into it.

@wolfv - do you think it would be possible to instead of installing libs/executables in the global `library/bin` folder instead install them in subdirectories with the name of the package (e.g. `library/bin/ros-galactic-demo-nodes-py`? In that case we would avoid those ClobberErrors.

Tobias-Fischer, 2021-08-30:
I think this is indeed what they do when packaging ROS for Windows: https://github.com/ros2/ros2/releases/download/release-galactic-20210716/ros2-galactic-20210616-windows-release-amd64.zip

Is that correct @ooeygui?

wolfv, 2021-08-30:
yeah, it would be possible to put them into different folders, but they would not be easily callable anymore. we could also prefix them, e.g. `talker.py -> demo-talker.py` etc.

traversaro, 2021-08-30:
> 
> 
> yeah, it would be possible to put them into different folders, but they would not be easily callable anymore. we could also prefix them, e.g. `talker.py -> demo-talker.py` etc.

I guess the basic idea is that those executables are meant only be called via `ros run`/`ros2 run`, but I may be wrong.

Tobias-Fischer, 2021-08-30:
I agree with @traversaro - as long as you can run the executable with `ros2 run` we should be fine, isn't it? I think https://github.com/ros2/ros2cli/blob/1f4a3aaf7b930830b3fa04773715bf93ea65693e/ros2pkg/ros2pkg/api/__init__.py#L40-L55 indicates that putting them in subdirectories called the same as the `package_name` should make them callable?

ooeygui, 2021-08-30:
I'm sorry I can't help on this one. The release you linked to is published by Open Robotics. The Foxy chocolatey package is composed during a build; last one wins. We don't see this clobber issue.

PrasadNR, 2021-08-31:
[image]
[image]

PrasadNR, 2021-08-31:
I was able to do `rosdep init` and `rosdep update`. But, when I do `ros2 run`, I see `'ros2' is not recognised as an internal or external command`.

Tobias-Fischer, 2021-08-31:
It seems like `ClobberErrror` actually is an error in this case, and not just a warning ;).

This is a relatively major issue then, as users can't easily install ros-galactic-desktop on Windows at the moment. Seeing that we are not 100% sure what the proper fix is (I guess subdirs of some sort ..), should we just patch the couple of packages involved as suggested by @wolfv and rename the affected files?

Tobias-Fischer, 2021-08-31:
Ok, I added some patches and hopefully this should resolve the problem. @PrasadNR could you please try again and let us know? (please wait at least an hour from now for the packages to be built)

Tobias-Fischer, 2021-08-31:
I just realised - are there more ClobberErrors that we can't see in the screenshot of the terminal window? If so, please copy+paste all of them (no screenshots, please ..).

PrasadNR, 2021-09-05:
Yeah. Sorry for the screenshot. I was using Windows after years of usage of Ubuntu. For some reason, in Anaconda prompt, I had to right click and then do "mark". In Windows, I do not even get an acknowledgement that the clipboard has the copied information. I think I will post any relevant information in the next hour.

PrasadNR, 2021-09-05:
Actually, I am simply copying and pasting the commands from the documentation https://github.com/RoboStack/ros-galactic. I think some command might be causing the anaconda to deactivate the environment itself. I am not sure why that is so.

I think (but I am not sure) it is this line that is causing problems like this.
`mamba install compilers cmake pkg-config make ninja colcon-common-extensions`
```

C:\Users\Prasad>conda activate ros_galactic

C:\Users\Prasad>set "LIB="

C:\Users\Prasad>set "CPATH="

C:\Users\Prasad>set "_LIB="

C:\Users\Prasad>set "_CPATH="

C:\Users\Prasad>if "C:\Users\Prasad\anaconda3\envs\ros_galactic" == "" set "LIB=
;\Library\lib"

C:\Users\Prasad>if "C:\Users\Prasad\anaconda3\envs\ros_galactic" == "" set "CPAT
H=;\Library\include"

C:\Users\Prasad>if not "C:\Users\Prasad\anaconda3\envs\ros_galactic" == "" set "
LIB=;C:\Users\Prasad\anaconda3\envs\ros_galactic\Library\lib"

C:\Users\Prasad>if not "C:\Users\Prasad\anaconda3\envs\ros_galactic" == "" set "
CPATH=;C:\Users\Prasad\anaconda3\envs\ros_galactic\Library\include"
)' was unexpected at this time.
C:\Users\Prasad>
```

I think I will try this later.

PrasadNR, 2021-09-06:
After this line `mamba install ros-galactic-desktop`, I see the following:
```
(ros_galactic) C:\Users\Prasad>conda deactivate

(base) C:\Users\Prasad>conda activate ros_galactic
)' was unexpected at this time.
C:\Users\Prasad>
```

Tobias-Fischer, 2021-09-06:
Can you just type `set` and report the output? Before and after `conda activate ros_galactic`? I am assuming you still use the Anaconda prompt? Can you try not using an underscore in the environment name? Can you try using `conda` instead of `mamba`?

PrasadNR, 2021-09-06:
```

(base) C:\Users\Prasad>set
ALLUSERSPROFILE=C:\ProgramData
APPDATA=C:\Users\Prasad\AppData\Roaming
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=LENOVO-PC
ComSpec=C:\windows\system32\cmd.exe
CONDA_DEFAULT_ENV=base
CONDA_EXE=C:\Users\Prasad\anaconda3\Scripts\conda.exe
CONDA_PREFIX=C:\Users\Prasad\anaconda3
CONDA_PROMPT_MODIFIER=(base)
CONDA_PYTHON_EXE=C:\Users\Prasad\anaconda3\python.exe
CONDA_SHLVL=1
configsetroot=C:\windows\ConfigSetRoot
easyplussdk="C:\Program Files (x86)\Common Files\lenovo\easyplussdk\bin"
FP_NO_HOST_CHECK=NO
HOMEDRIVE=C:
HOMEPATH=\Users\Prasad
LOCALAPPDATA=C:\Users\Prasad\AppData\Local
LOGONSERVER=\\LENOVO-PC
NUMBER_OF_PROCESSORS=8
OS=Windows_NT
Path=C:\Users\Prasad\anaconda3;C:\Users\Prasad\anaconda3\Library\mingw-w64\bin;C
:\Users\Prasad\anaconda3\Library\usr\bin;C:\Users\Prasad\anaconda3\Library\bin;C
:\Users\Prasad\anaconda3\Scripts;C:\Users\Prasad\anaconda3\bin;C:\Users\Prasad\a
naconda3\condabin;C:\Program Files (x86)\Lenovo\FusionEngine;C:\Program Files (x
86)\NVIDIA Corporation\PhysX\Common;C:\Program Files (x86)\Intel\iCLS Client;C:\
Program Files\Intel\iCLS Client;C:\windows\system32;C:\windows;C:\windows\System
32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0;C:\Program Files\Intel\Intel(
R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management E
ngine Components\IPT;C:\Program Files (x86)\Intel\Intel(R) Management Engine Com
ponents\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\I
PT;C:\Program Files (x86)\Lenovo\Motion Control;C:\Program Files (x86)\Common Fi
les\lenovo\easyplussdk\bin;C:\Program Files\PuTTY;C:\Users\Prasad\AppData\Local\
Programs\Python\Python39\Scripts;C:\Users\Prasad\AppData\Local\Programs\Python\P
ython39
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 60 Stepping 3, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=3c03
ProgramData=C:\ProgramData
ProgramFiles=C:\Program Files
ProgramFiles(x86)=C:\Program Files (x86)
ProgramW6432=C:\Program Files
PROMPT=(base) $P$G
PSModulePath=C:\windows\system32\WindowsPowerShell\v1.0\Modules\
PUBLIC=C:\Users\Public
SESSIONNAME=Console
SystemDrive=C:
SystemRoot=C:\windows
TEMP=C:\Users\Prasad\AppData\Local\Temp
TMP=C:\Users\Prasad\AppData\Local\Temp
USERDOMAIN=Lenovo-PC
USERDOMAIN_ROAMINGPROFILE=Lenovo-PC
USERNAME=Prasad
USERPROFILE=C:\Users\Prasad
windir=C:\windows

(base) C:\Users\Prasad>conda activate ros_galactic

(ros_galactic) C:\Users\Prasad>set
ALLUSERSPROFILE=C:\ProgramData
AMENT_PREFIX_PATH=C:\Users\Prasad\anaconda3\envs\ros_galactic\Library
ament_python_executable=)=^)
APPDATA=C:\Users\Prasad\AppData\Roaming
CommonProgramFiles=C:\Program Files\Common Files
CommonProgramFiles(x86)=C:\Program Files (x86)\Common Files
CommonProgramW6432=C:\Program Files\Common Files
COMPUTERNAME=LENOVO-PC
ComSpec=C:\windows\system32\cmd.exe
CONDA_DEFAULT_ENV=ros_galactic
CONDA_EXE=C:\Users\Prasad\anaconda3\Scripts\conda.exe
CONDA_PREFIX=C:\Users\Prasad\anaconda3\envs\ros_galactic
CONDA_PREFIX_1=C:\Users\Prasad\anaconda3
CONDA_PROMPT_MODIFIER=(ros_galactic)
CONDA_PYTHON_EXE=C:\Users\Prasad\anaconda3\python.exe
CONDA_SHLVL=2
configsetroot=C:\windows\ConfigSetRoot
easyplussdk="C:\Program Files (x86)\Common Files\lenovo\easyplussdk\bin"
FP_NO_HOST_CHECK=NO
HOMEDRIVE=C:
HOMEPATH=\Users\Prasad
LOCALAPPDATA=C:\Users\Prasad\AppData\Local
LOGONSERVER=\\LENOVO-PC
NUMBER_OF_PROCESSORS=8
OS=Windows_NT
Path=C:\Users\Prasad\anaconda3\envs\ros_galactic;C:\Users\Prasad\anaconda3\envs\
ros_galactic\Library\mingw-w64\bin;C:\Users\Prasad\anaconda3\envs\ros_galactic\L
ibrary\usr\bin;C:\Users\Prasad\anaconda3\envs\ros_galactic\Library\bin;C:\Users\
Prasad\anaconda3\envs\ros_galactic\Scripts;C:\Users\Prasad\anaconda3\envs\ros_ga
lactic\bin;C:\Users\Prasad\anaconda3\condabin;C:\Program Files (x86)\Lenovo\Fusi
onEngine;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files
 (x86)\Intel\iCLS Client;C:\Program Files\Intel\iCLS Client;C:\windows\system32;
C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0;C
:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files
\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\Intel\In
tel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Ma
nagement Engine Components\IPT;C:\Program Files (x86)\Lenovo\Motion Control;C:\P
rogram Files (x86)\Common Files\lenovo\easyplussdk\bin;C:\Program Files\PuTTY;C:
\Users\Prasad\AppData\Local\Programs\Python\Python39\Scripts;C:\Users\Prasad\App
Data\Local\Programs\Python\Python39
PATHEXT=.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
PROCESSOR_ARCHITECTURE=AMD64
PROCESSOR_IDENTIFIER=Intel64 Family 6 Model 60 Stepping 3, GenuineIntel
PROCESSOR_LEVEL=6
PROCESSOR_REVISION=3c03
ProgramData=C:\ProgramData
ProgramFiles=C:\Program Files
ProgramFiles(x86)=C:\Program Files (x86)
ProgramW6432=C:\Program Files
PROJ_LIB=C:\Users\Prasad\anaconda3\envs\ros_galactic\Library\share\proj
PROJ_NETWORK=ON
PROMPT=(ros_galactic) $P$G
PSModulePath=C:\windows\system32\WindowsPowerShell\v1.0\Modules\
PUBLIC=C:\Users\Public
PYTHONPATH=C:\Users\Prasad\anaconda3\envs\ros_galactic\Library\lib/site-packages
;C:\Users\Prasad\anaconda3\envs\ros_galactic\Library\Lib\site-packages
QT_PLUGIN_PATH=C:\Users\Prasad\anaconda3\envs\ros_galactic\Library\plugins
ROS_DISTRO=galactic
ROS_ETC_DIR=C:\Users\Prasad\anaconda3\envs\ros_galactic\Library\etc\ros
ROS_LOCALHOST_ONLY=0
ROS_OS_OVERRIDE=robostack:win64
ROS_PYTHON_VERSION=3
ROS_VERSION=2
SESSIONNAME=Console
SystemDrive=C:
SystemRoot=C:\windows
TEMP=C:\Users\Prasad\AppData\Local\Temp
TMP=C:\Users\Prasad\AppData\Local\Temp
USERDOMAIN=Lenovo-PC
USERDOMAIN_ROAMINGPROFILE=Lenovo-PC
USERNAME=Prasad
USERPROFILE=C:\Users\Prasad
windir=C:\windows

(ros_galactic) C:\Users\Prasad>
```

PrasadNR, 2021-09-06:
```
conda create -n rosgalactic python=3.8
conda activate rosgalactic
conda config --env --add channels conda-forge
conda config --env --add channels robostack-experimental
conda config --env --set channel_priority strict
conda install ros-galactic-desktop
conda deactivate
```
```
(base) C:\Users\Prasad>conda activate rosgalactic
)' was unexpected at this time.
C:\Users\Prasad>
```

Tobias-Fischer, 2021-09-06:
I'm not sure what's going on with this `)' was unexpected at this time.` error. Unfortunately I'm not a Windows expert - any clues @wolfv @traversaro?

traversaro, 2021-09-06:
I was able to reproduce the issue.

traversaro, 2021-09-06:
I opened a new issue to track this as it seems a different issue from the one originally described in this issue: https://github.com/RoboStack/ros-galactic/issues/51 .

traversaro, 2021-09-06:
@PrasadNR as discussed in https://github.com/RoboStack/ros-galactic/issues/51, a workaround is just to re-activate the environment in a new command prompt.

PrasadNR, 2021-09-06:
> @PrasadNR as discussed in #51, a workaround is just to re-activate the environment in a new command prompt.

Awesome. Now I do not see `ClobberError`s.

closed 2021-09-06 by PrasadNR