RoboStack/ros-galactic #51: )' was unexpected at this time. error on activate on Windows Command Prompt

traversaro, 2021-09-06:
Originally reported by @PrasadNR in https://github.com/RoboStack/ros-galactic/issues/50#issuecomment-913289398 : 
> 
> 
> ```
> conda create -n rosgalactic python=3.8
> conda activate rosgalactic
> conda config --env --add channels conda-forge
> conda config --env --add channels robostack-experimental
> conda config --env --set channel_priority strict
> conda install ros-galactic-desktop
> conda deactivate
> ```
> 
> ```
> (base) C:\Users\Prasad>conda activate rosgalactic
> )' was unexpected at this time.
> C:\Users\Prasad>
> ```

I was able to reproduce the issue. I tracked it to the `%CONDA_PREFIX%/Library/local_setup.bat` script. I added a few debug prints, in:
~~~
  echo "Call local_setup.bat 01"

  :: check environment variable for custom Python executable
  if "%AMENT_PYTHON_EXECUTABLE%" NEQ "" (
    echo "Call local_setup.bat 02"
    if not exist "%AMENT_PYTHON_EXECUTABLE%" (
      echo error: AMENT_PYTHON_EXECUTABLE '%AMENT_PYTHON_EXECUTABLE%' doesn't exist
      exit /b 1
    )
    set "_ament_python_executable=%AMENT_PYTHON_EXECUTABLE%"
  ) else (
    echo "Call local_setup.bat 03"
    :: use the Python executable known at configure time
    set "_ament_python_executable=C:/Users/STraversaro/AppData/Local/mambaforge/envs/rosgalactic/python.exe"
    :: if it doesn't exist try a fall back
    if not exist "!_ament_python_executable!" (
      python --version > NUL 2> NUL
      if errorlevel 1 (
        echo error: unable to find python executable
        exit /b 1
      )
      set "_ament_python_executable=python"
    )
  )

  echo "Call local_setup.bat 08"
~~~

And this is the output:
~~~
C:\Users\STraversaro\AppData\Local\mambaforge\envs\rosgalactic\etc\conda\activate.d>conda activate rosgalactic

C:\Users\STraversaro\AppData\Local\mambaforge\envs\rosgalactic\etc\conda\activate.d>echo "Call activate proj4"
"Call activate proj4"

C:\Users\STraversaro\AppData\Local\mambaforge\envs\rosgalactic\etc\conda\activate.d>echo "End activate proj4"
"End activate proj4"

C:\Users\STraversaro\AppData\Local\mambaforge\envs\rosgalactic\etc\conda\activate.d>echo "Call activate rosgalatic 1"
"Call activate rosgalatic 1"

C:\Users\STraversaro\AppData\Local\mambaforge\envs\rosgalactic\etc\conda\activate.d>echo "Call activate rosgalatic 2"
"Call activate rosgalatic 2"

C:\Users\STraversaro\AppData\Local\mambaforge\envs\rosgalactic\etc\conda\activate.d>echo "Call activate rosgalatic 3"
"Call activate rosgalatic 3"
"Call local_setup.bat 0"
"Call local_setup.bat 01"
)' was unexpected at this time.
C:\Users\STraversaro\AppData\Local\mambaforge\envs\rosgalactic\etc\conda\activate.d>
~~~

wolfv, 2021-09-06:
were you able to find the offending `.bat` file? Looks like some wrongly inserted characters ...

traversaro, 2021-09-06:
It seems to be `%CONDA_PREFIX%/Library/local_setup.bat` .

traversaro, 2021-09-06:
The error is that the `echo` commands to not have quotes. If that is fixed the error that is printed is:
~~~
(rosgalactic) C:\Users\STraversaro\AppData\Local\mambaforge\envs\rosgalactic\etc\conda\activate.d>conda activate rosgalactic
"error: AMENT_PYTHON_EXECUTABLE '^^^)=^^^^)' doesn't exist"
~~~

traversaro, 2021-09-06:
Indeed, the strange value of `AMENT_PYTHON_EXECUTABLE` can be found by just inspecting the env variable set:
~~~
(rosgalactic) C:\Users\STraversaro\AppData\Local\mambaforge\envs\rosgalactic\etc\conda\activate.d>set
ALLUSERSPROFILE=C:\ProgramData
AMENT_PREFIX_PATH=C:\Users\STraversaro\AppData\Local\mambaforge\envs\rosgalactic\Library
ament_python_executable=^^^)=^^^^)
APPDATA=C:\Users\STraversaro\AppData\Roaming
CommonProgramFiles=C:\Program Files\Common Files
~~~

traversaro, 2021-09-06:
Note that this is only happening if you try to run `conda activate rosgalactic` in the same command prompt in which you previously activate and de-activated `rosgalactic` environment, i.e. in a new prompt:
~~~
conda activate rosgalactic
~~~
works fine, while:
~~~
C:\Users\STraversaro>conda activate rosgalactic

(rosgalactic) C:\Users\STraversaro>conda deactivate

C:\Users\STraversaro>conda activate rosgalactic
"error: AMENT_PYTHON_EXECUTABLE ')=^)' doesn't exist"
~~~

Fails.

traversaro, 2021-09-06:
> The error is that the `echo` commands to not have quotes. If that is fixed the error that is printed is:

Note that this happens only because `AMENT_PREFIX_PATH` contains parenthesis, otherwise the error message works fine, see:
~~~
C:\Users\STraversaro>set
ALLUSERSPROFILE=C:\ProgramData
ament_python_executable=)=^)
APPDATA=C:\Users\STraversaro\AppData\Roaming

C:\Users\STraversaro>conda activate rosgalactic
)' was unexpected at this time.
C:\Users\STraversaro>set ament_python_executable=wrongstringthatdoesnothaveparenthesis

C:\Users\STraversaro>conda activate rosgalactic
error: AMENT_PYTHON_EXECUTABLE 'wrongstringthatdoesnothaveparenthesis' doesn't exist

(rosgalactic) C:\Users\STraversaro>
~~~

traversaro, 2021-09-06:
Related issue: https://github.com/ament/ament_package/issues/84 .

traversaro, 2021-09-06:
This is caused by this two (different) issues: 
* https://github.com/RoboStack/ros-galactic/issues/52
* https://github.com/RoboStack/ros-galactic/issues/53

Furthermore, the error was quite criptic due to the problem in printing errors if `AMENT_PYTHON_EXECUTABLE` contained a parenthesis.

PrasadNR, 2021-09-06:
```

(base) C:\Users\Prasad>conda activate rosgalactic

(rosgalactic) C:\Users\Prasad>conda install compilers cmake pkg-config make ninj
a colcon-common-extensions
Collecting package metadata (current_repodata.json): done
Solving environment: failed with initial frozen solve. Retrying with flexible so
lve.
Solving environment: failed with repodata from current_repodata.json, will retry
 with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: done

## Package Plan ##

  environment location: C:\Users\Prasad\anaconda3\envs\rosgalactic

  added / updated specs:
    - cmake
    - colcon-common-extensions
    - compilers
    - make
    - ninja
    - pkg-config

The following packages will be downloaded:

    package                    |            build
    ---------------------------|-----------------
    c-compiler-1.3.0           |       h8ffe710_0           5 KB  conda-forge
    clangdev-5.0.0             |          flang_3       189.0 MB  conda-forge
    compilers-1.3.0            |       h57928b3_0           5 KB  conda-forge
    cxx-compiler-1.3.0         |       h2d74725_0           5 KB  conda-forge
    fortran-compiler-1.3.0     |       h9655429_0           5 KB  conda-forge
    ------------------------------------------------------------
                                           Total:       189.0 MB

The following NEW packages will be INSTALLED:

  c-compiler         conda-forge/win-64::c-compiler-1.3.0-h8ffe710_0
  clangdev           conda-forge/win-64::clangdev-5.0.0-flang_3
  colcon-cmake       conda-forge/noarch::colcon-cmake-0.2.26-py_0
  colcon-common-ext~ conda-forge/win-64::colcon-common-extensions-0.2.1-py38haa2
44fe_2
  colcon-core        conda-forge/win-64::colcon-core-0.6.1-py38haa244fe_1
  colcon-defaults    conda-forge/noarch::colcon-defaults-0.2.5-py_0
  colcon-devtools    conda-forge/noarch::colcon-devtools-0.2.2-py_0
  colcon-library-pa~ conda-forge/noarch::colcon-library-path-0.2.1-py_0
  colcon-metadata    conda-forge/noarch::colcon-metadata-0.2.5-py_0
  colcon-notificati~ conda-forge/win-64::colcon-notification-0.2.13-py38h32f6830
_1
  colcon-output      conda-forge/noarch::colcon-output-0.2.12-py_0
  colcon-package-in~ conda-forge/noarch::colcon-package-information-0.3.3-py_0
  colcon-package-se~ conda-forge/noarch::colcon-package-selection-0.2.10-py_0
  colcon-parallel-e~ conda-forge/noarch::colcon-parallel-executor-0.2.4-py_0
  colcon-pkg-config  conda-forge/noarch::colcon-pkg-config-0.1.0-py_0
  colcon-powershell  conda-forge/noarch::colcon-powershell-0.3.6-py_0
  colcon-python-set~ conda-forge/noarch::colcon-python-setup-py-0.2.7-py_0
  colcon-recursive-~ conda-forge/noarch::colcon-recursive-crawl-0.2.1-py_0
  colcon-ros         conda-forge/noarch::colcon-ros-0.3.21-py_0
  colcon-test-result conda-forge/noarch::colcon-test-result-0.3.8-py_0
  coloredlogs        conda-forge/win-64::coloredlogs-15.0.1-py38haa244fe_1
  compilers          conda-forge/win-64::compilers-1.3.0-h57928b3_0
  coverage           conda-forge/win-64::coverage-5.5-py38h294d835_0
  cxx-compiler       conda-forge/win-64::cxx-compiler-1.3.0-h2d74725_0
  distlib            conda-forge/noarch::distlib-0.3.2-pyhd8ed1ab_0
  flang              conda-forge/win-64::flang-5.0.0-he025d50_20180525
  flang_win-64       conda-forge/win-64::flang_win-64-5.0.0-h13ae965_20180526
  fortran-compiler   conda-forge/win-64::fortran-compiler-1.3.0-h9655429_0
  humanfriendly      conda-forge/win-64::humanfriendly-9.2-py38haa244fe_1
  libflang           conda-forge/win-64::libflang-5.0.0-h6538335_20180525
  llvm-meta          conda-forge/noarch::llvm-meta-5.0.0-0
  make               conda-forge/win-64::make-4.3-h3d2af85_1
  ninja              conda-forge/win-64::ninja-1.10.2-h5362a0b_0
  openmp             conda-forge/win-64::openmp-5.0.0-vc14_1
  pyreadline         conda-forge/win-64::pyreadline-2.1-py38haa244fe_1003
  pytest-cov         conda-forge/noarch::pytest-cov-2.12.1-pyhd8ed1ab_0
  pytest-repeat      conda-forge/noarch::pytest-repeat-0.8.0-py_0
  pytest-rerunfailu~ conda-forge/noarch::pytest-rerunfailures-10.1-pyhd8ed1ab_0
  pytest-runner      conda-forge/noarch::pytest-runner-5.3.1-pyhd8ed1ab_0
  pywin32            conda-forge/win-64::pywin32-301-py38h294d835_0
  vs2017_win-64      conda-forge/win-64::vs2017_win-64-19.16.27038-h2e3bad8_2
  vswhere            conda-forge/win-64::vswhere-2.8.4-h57928b3_0

Proceed ([y]/n)? y

Downloading and Extracting Packages
clangdev-5.0.0       | 189.0 MB  | #################################### | 100%
fortran-compiler-1.3 | 5 KB      | #################################### | 100%
cxx-compiler-1.3.0   | 5 KB      | #################################### | 100%
c-compiler-1.3.0     | 5 KB      | #################################### | 100%
compilers-1.3.0      | 5 KB      | #################################### | 100%
Preparing transaction: done
Verifying transaction: done
Executing transaction: done

C:\Users\Prasad>set "LIB="

C:\Users\Prasad>set "CPATH="

C:\Users\Prasad>set "_LIB="

C:\Users\Prasad>set "_CPATH="

C:\Users\Prasad>if "C:\Users\Prasad\anaconda3\envs\rosgalactic" == "" set "LIB=;
\Library\lib"

C:\Users\Prasad>if "C:\Users\Prasad\anaconda3\envs\rosgalactic" == "" set "CPATH
=;\Library\include"

C:\Users\Prasad>if not "C:\Users\Prasad\anaconda3\envs\rosgalactic" == "" set "L
IB=;C:\Users\Prasad\anaconda3\envs\rosgalactic\Library\lib"

C:\Users\Prasad>if not "C:\Users\Prasad\anaconda3\envs\rosgalactic" == "" set "C
PATH=;C:\Users\Prasad\anaconda3\envs\rosgalactic\Library\include"
)' was unexpected at this time.
C:\Users\Prasad>
```

Tobias-Fischer, 2021-09-20:
Should we close this? I think half of the issue has been fixed, and the other half is tracked in #53

traversaro, 2021-09-20:
I agree.

closed 2021-09-20 by traversaro