RoboStack/ros-galactic #53: ament_python_executable

traversaro, 2021-09-06:


traversaro, 2021-09-06:
It seems that something wrong is happening in https://github.com/ament/ament_package/blob/b3669d80d8166f14871d98555dd1d5083e35ecde/ament_package/template/prefix_level/local_setup.bat.in#L23, that gets installed in `%CONDA_PREFIX%/Library/local_setup.bat` .

Tobias-Fischer, 2021-09-06:
Should we just set `AMENT_PYTHON_EXECUTABLE` in https://github.com/RoboStack/vinca/blob/master/vinca/templates/activate.bat.in (before line https://github.com/RoboStack/vinca/blob/54dc6b51d9b3e4c3a307ede50b37c406e0a5547e/vinca/templates/activate.bat.in#L10)? Not sure what it needs to be set to in Windows though.

Tobias-Fischer, 2021-09-20:
@traversaro - any clues what `AMENT_PYTHON_EXECUTABLE` should point to?

traversaro, 2021-09-20:
The `%CONDA_PREFIX%/Library/local_setup.bat` file contains the line: 
~~~
     :: use the Python executable known at configure time
    set "_ament_python_executable=C:\Users\STraversaro\AppData\Local\mambaforge\envs\rosgalactic\python.exe"
~~~

That for some reason is not working (even if it is executed). However, this seems to indicate that `AMENT_PYTHON_EXECUTABLE` should point to `%CONDA_PREFIX%\Library\python.exe` on Windows.

Tobias-Fischer, 2021-09-20:
Fixed in https://github.com/RoboStack/vinca/commit/204c8e3c55ca7333290fd6938cdab5cfaefdc226

Needs rebuild of the ROS packages though to take effect.

@wolfv - shall we tackle this soon?

Tobias-Fischer, 2021-09-20:
Thanks @traversaro by the way :)

closed 2021-09-28 by Tobias-Fischer

danzimmerman, 2022-06-23:
> However, this seems to indicate that AMENT_PYTHON_EXECUTABLE should point to %CONDA_PREFIX%\Library\python.exe on Windows.

I don't have a `python.exe` in `Library` (that folder is such a pain in Windows Conda 😓). 

I'm getting this trying to activate the `setup.bat` from my Colcon workspace

```
(galactic-clean) C:\Code\ros2\galactic_ws>install\setup.bat
error: AMENT_PYTHON_EXECUTABLE 'C:\Users\dan\mambaforge\envs\galactic-clean\Library\python.exe' doesn't exist
```

This is happy enough if I set it back to `envs/galactic-clean/python.exe` with:

```
(galactic-clean) C:\Code\ros2\galactic_ws>set AMENT_PYTHON_EXECUTABLE=%CONDA_PREFIX%\python.exe

(galactic-clean) C:\Code\ros2\galactic_ws>install\setup.bat

(galactic-clean) C:\Code\ros2\galactic_ws>
```
I'm getting problems launching things after that, so I'm not so sure it's right either (I don't really know what `AMENT_PYTHON_EXECUTABLE` is for yet, will update)

PR #54 (2021-09-28): Global rebuild
Fixes #53 #39 

Needs #48 and https://github.com/RoboStack/vinca/pull/18 first 

Will enable build of #32