RoboStack/ros-galactic #70: Fix libyaml-vendor

Tobias-Fischer, 2022-01-09:
At the moment we use a yaml build that is incompatible with the one on conda-forge (https://anaconda.org/robostack-experimental/yaml/files) [just for osx-arm64]. We need to fix libyaml-vendor to deal with pkg-config files rather than just cmake files, as that is not available on conda-forge.

traversaro, 2022-01-17:
`libyaml_vendor` at the moment is not even searching for `libyaml` in the system. This should be fixed by https://github.com/ros2/libyaml_vendor/pull/41 .

Tobias-Fischer, 2022-02-02:
Hi @traversaro - did you by any chance have a look at this?

Tobias-Fischer, 2022-02-02:
By the way, we already patch in https://github.com/RoboStack/ros-galactic/blob/main/patch/ros-galactic-libyaml-vendor.patch

traversaro, 2022-02-02:
No, I need to look into this in the next days. Thanks for mentioning the existing patch, I did not see that before.

closed 2022-02-06 by Tobias-Fischer