RoboStack/ros-galactic #72: Fix mimick-vendor cross-compile for osx-arm64

Tobias-Fischer, 2022-01-16:
mimick-vendor currently only compiles natively - cross compile doesn't work.

Tobias-Fischer, 2022-01-18:
Fixed the mimick-vendor cross-compile; qt-gui-cpp seems trickier.

michtesar, 2022-01-18:
I am investigating `qt-gui-cpp`.

michtesar, 2022-01-18:
So far it seems I am missing a dependencies to even start building `qt-gui-package`.

```
Mamba failed to solve:
 - clangxx_osx-arm64 12*
 - clang_osx-arm64 12*
 - ninja
 - make
 - tapi
 - cmake
 - python 3.8.* *_cpython
 - cross-python_osx-arm64
 - cython
 - pkg-config
 - pyqt-builder

with channels:

The reported errors are:
- Encountered problems while solving:
-   - nothing provides requested pyqt-builder
```

michtesar, 2022-01-18:
After some testing, I tried to compile `qt-gui-cpp` for `osx-64` and `osx-arm64` on Intel based MacBook Pro.

In `recipe.yaml` I added followings host libraries:

```
- qt-main
- pyqt-builder
- libfido2
- qt
```

I found issues with incompatibility of versions of `qt`. During build I see `package qt-main-5.15.2-hfad5ee4_0 has constraint qt 5.15.2 conflicting with qt-5.12.1-h1b46049_0`.

For cross-platform build for `osx-arm64` it goes until almost end and complains on C lang compiler as manifested as follows:

```
ld: library not found for -lSystem
    clang-12: error: linker command failed with exit code 1 (use -v to see invocation)
    ninja: build stopped: subcommand failed.
    
  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)
```

Its noteworthy that in `recipe.yaml` for `osx-arm64` I commented out `qt` package otherwise, there are no sources to `qt 5.12.*`.

I include log file for both to start investigating.

[report_arm-x64.log](https://github.com/RoboStack/ros-galactic/files/7889645/report_arm-x64.log)
[report_x64.log](https://github.com/RoboStack/ros-galactic/files/7889646/report_x64.log)

michtesar, 2022-01-18:
It seems I can build it on `osx-64`, with `osx-arm64` I struggle with qt versions.

Tobias-Fischer, 2022-01-18:
Good start!

I forgot to tell you one piece that explains the `qt` issues. For the osx-arm64 build, we only use the `qt-main` package. This is already configured properly in https://github.com/RoboStack/ros-galactic/blob/main/robostack.yaml - however, you need to tell vinca that you want to create recipes for osx-arm64 rather than osx-64. You do so by running `vinca --multiple --platform=osx-arm64`. Then `qt` should disappear in the `recipe.yaml`.

The error `ld: library not found for -lSystem` can be explained as `CONDA_BUILD_SYSROOT` needs to be set on your machine: https://github.com/RoboStack/ros-galactic/blob/5dd8f8992639e315be47d96116d2d0bb8e310fd1/conda_build_config.yaml#L64-L65 - simply add a new entry with the path to your SDK. You can download the 11.3 SDK from https://github.com/phracker/MacOSX-SDKs/releases/tag/11.3 and point to that path.

Let us know how you go.