RoboStack/ros-galactic #73: Fix osx-arm64 (M1) codesign issues

Tobias-Fischer, 2022-01-19:
We now have builds for packages up to `ros-galactic-desktop` for osx-am64. However, to make things work, after installing/updating packages, one needs to run:

```
find $CONDA_PREFIX -name "*.so" -exec codesign -s - -f {} \;
find $CONDA_PREFIX -name "*.dylib" -exec codesign -s - -f {} \;
```

As the signatures are not valid. 

We are not sure whether this is an issue in RoboStack, Conda-forge, mamba, conda, ... so it's a bit hard to track down.

/cc @wolfv

Tobias-Fischer, 2022-02-02:
Might be because of the newer compiler version that I used. It does not happen in ros-noetic where the old compiler is used. Hopefully fixed by #75

KohlhardtC, 2022-02-24:
Running the command above did not work when the ros_galactic environment was active. The fix is to run `conda deactivate` before running the above commands to update the signatures.

Tobias-Fischer, 2022-04-07:
This is not an issue anymore with the Python 3.9 builds :)

closed 2022-04-07 by Tobias-Fischer