RoboStack/ros-galactic #74: Nodes from demo_nodes_py package can't be started via ros2 run on Windows

traversaro, 2022-01-20:
On my Windows system, if I try to run:
~~~
ros2 run demo_nodes_py demo_talker
~~~
it fails with:
~~~
No executable found
~~~

Other similar nodes instead work fine, for example:
~~~
(ros2) C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2>ros2 run demo_nodes_cpp talker
~~~
whose output is:
~~~
[INFO] [1642666133.989059100] [talker]: Publishing: 'Hello World: 1'
[INFO] [1642666134.978791100] [talker]: Publishing: 'Hello World: 2'
~~~

<br/>
Environment (<code>conda list</code>):
<details>

```
$ conda list
(ros2) C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2>conda list
# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2:
#
# Name                    Version                   Build  Channel
argcomplete               2.0.0              pyhd8ed1ab_0    conda-forge
assimp                    5.0.1                hc2aa0de_7    conda-forge
atomicwrites              1.4.0              pyh9f0ad1d_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
boost                     1.74.0           py38he5193b3_4    conda-forge
boost-cpp                 1.74.0               h5b4e17d_5    conda-forge
brotli                    1.0.9                h8ffe710_6    conda-forge
brotli-bin                1.0.9                h8ffe710_6    conda-forge
bullet                    3.17                 h57928b3_1    conda-forge
bullet-cpp                3.17                 h5d928e2_1    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
cairo                     1.16.0            hb19e0ff_1008    conda-forge
catkin_pkg                0.4.24             pyhd8ed1ab_1    conda-forge
certifi                   2021.10.8        py38haa244fe_1    conda-forge
cffi                      1.15.0           py38hd8c33c5_0    conda-forge
cmake                     3.21.3               h39d44d4_0    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
console_bridge            1.0.1                h5362a0b_0    conda-forge
cppcheck                  2.6.2            py38h1c6bfe0_0    conda-forge
cryptography              36.0.1           py38hb7941b4_0    conda-forge
curl                      7.81.0               h789b8ee_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.18.1           py38haa244fe_0    conda-forge
double-conversion         3.1.7                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.3                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flake8                    4.0.1              pyhd8ed1ab_0    conda-forge
flann                     1.9.1             h7dfa0ef_1008    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.1            h1989441_1005    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.28.5           py38h294d835_0    conda-forge
foonathan-memory          0.6.2                h57928b3_2    conda-forge
[... 454 lines cut ...]
                          C:\Users\STraversaro\.condarc
          conda version : 4.10.3
    conda-build version : not installed
         python version : 3.9.7.final.0
       virtual packages : __cuda=11.6=0
                          __win=0=0
                          __archspec=1=x86_64
       base environment : C:\Users\STraversaro\AppData\Local\mambaforge  (writable)
      conda av data dir : C:\Users\STraversaro\AppData\Local\mambaforge\etc\conda
  conda av metadata url : None
           channel URLs : https://conda.anaconda.org/conda-forge/win-64
                          https://conda.anaconda.org/conda-forge/noarch
          package cache : C:\Users\STraversaro\AppData\Local\mambaforge\pkgs
                          C:\Users\STraversaro\.conda\pkgs
                          C:\Users\STraversaro\AppData\Local\conda\conda\pkgs
       envs directories : C:\Users\STraversaro\AppData\Local\mambaforge\envs
                          C:\Users\STraversaro\.conda\envs
                          C:\Users\STraversaro\AppData\Local\conda\conda\envs
               platform : win-64
             user-agent : conda/4.10.3 requests/2.26.0 CPython/3.9.7 Windows/10 Windows/10.0.22000
          administrator : False
             netrc file : None
           offline mode : False
```
</details>

traversaro, 2022-01-20:
The main problem is that the executable/scripts of [`ros-galactic-demo-nodes-py`](https://anaconda.org/robostack-experimental/ros-galactic-demo-nodes-py) packages get installed in `%CONDA_PREFIX%\Library\bin`, meaning that they can be executed directly, i.e.:
~~~
demo_talker
~~~
works, but they are not found by `ros2 run demo_nodes_py <executable>` as this command would searches for the commands in `%CONDA_PREFIX%\Library\lib` (see https://github.com/ros2/ros2cli/blob/0.16.1/ros2pkg/ros2pkg/api/__init__.py#L44).

So the problem seems related to the installation process of `demo_nodes_py` and why it install the scripts in the wrong location.

traversaro, 2022-01-20:
Interestingly, on Linux and macOS the scripts are installed in the correct location, i.e. `$CONDA_PREFIX/lib/demo_nodes_py`.

wolfv, 2022-01-20:
Yeah, Windows is unfortunately "special" in that the "unix" prefix is in `$CONDA_PREFIX\Library\...` 

The `Library\bin` folder should regularly be on the %PATH% (when activate -> https://github.com/mamba-org/mamba/blob/d4acbf7dc3d4743c805e8b9085033703f823b9d6/libmamba/src/core/activation.cpp#L180-L185) but maybe ROS has a special way of looking for the executables?

traversaro, 2022-01-20:
> Yeah, Windows is unfortunately "special" in that the "unix" prefix is in `$CONDA_PREFIX\Library\...`

The problem is not the usual special `<install_prefix>` on Windows. The issue is that on Linux/macOS the executables/scripts are installed in `<install_prefix>/lib/demo_nodes_py` (correct location), while on Windows they are in `<install_prefix>/bin` (wrong location). This is not directly related to what `<install_prefix>` is on Windows or *nix. These kind of executables are intentionally meant not be on the path, and to be accessible only via `ros2 run`, that looks in `<install_prefix>/lib/<pkg_name>`.

traversaro, 2022-01-20:
By the way, I just tried to build via colcon build the latest commit of `demo_nodes_py` package on Windows, and indeed they executables are correctly installed in `<install_prefix>/lib/<pkg_name>`.

Tobias-Fischer, 2022-01-20:
Hm is there anything wrong in https://github.com/RoboStack/vinca/blob/master/vinca/templates/bld_ament_python.bat.in?

Tobias-Fischer, 2022-01-20:
Also, does it just happen for python packages, or also for C++ packages (e.g. demo_nodes_cpp)?

traversaro, 2022-01-20:
> Also, does it just happen for python packages, or also for C++ packages (e.g. demo_nodes_cpp)?

C++ packages are fine, `ros2 run demo_nodes_cpp talker` works fine and you can check manually that the executables are installed in `<install_prefix>/lib/demo_nodes_cpp`.

traversaro, 2022-01-20:
> Hm is there anything wrong in https://github.com/RoboStack/vinca/blob/master/vinca/templates/bld_ament_python.bat.in?

The `--install-scripts=` option indeed seems to be quite related. By the way, the `.sh` variant of the same script seems to be quite different: https://github.com/RoboStack/vinca/blob/6777295bc3c140abf0d8ba518b9123050c5c28df/vinca/templates/build_ament_python.sh.in .

Tobias-Fischer, 2022-01-20:
Hm we could just try building the demo-nodes-py package with the same `pip` command as in the sh .. ?

Tobias-Fischer, 2022-01-20:
I rebuilt the demo-nodes-py package using the `pip` version: https://anaconda.org/robostack-experimental/ros-galactic-demo-nodes-py/0.14.3/download/win-64/ros-galactic-demo-nodes-py-0.14.3-py38hae75d3f_1.tar.bz2

Does that look better? (by the way, sorry, I overwrote the previous version ..)

Tobias-Fischer, 2022-01-21:
Looks like the executables are now in Library\lib :)

traversaro, 2022-01-21:
I just tested, and I still get:
~~~
(ros2ws4) C:\>ros2 run demo_nodes_py demo_talker
No executable found
~~~

> Looks like the executables are now in Library\lib :)

Ah, I see. The problem is that the executables should be in `Library\lib\demo_nodes_py` . Perhaps https://github.com/RoboStack/vinca/pull/28 could help?

Tobias-Fischer, 2022-01-21:
Reopening here until we know it's actually fixed

traversaro, 2022-01-21:
Ops! Now executables are installed in `Library\lib\ros-galactic-demo-nodes-py`, but they should be installed `Library\lib\demo_nodes_py`.

Tobias-Fischer, 2022-01-22:
Looks like it's working now :)

Tobias-Fischer, 2022-01-22:
Still leaving it open as it needs a rebuild of probably all windows packages.

traversaro, 2022-01-22:
I double checked and not indeed it works fine:
~~~
(ros2) C:\Users\STraversaro>ros2 run demo_nodes_py demo_talker
[INFO] [1642849699.319292400] [talker]: Publishing: "Hello World: 0"
[INFO] [1642849700.186619500] [talker]: Publishing: "Hello World: 1"
[INFO] [1642849701.185644300] [talker]: Publishing: "Hello World: 2"
~~~

Tobias-Fischer, 2022-01-26:
Hmm .. now we are running into new issues because of this:
```
CMake Error at C:/bld/ros-galactic-foonathan-memory-vendor_1643229794134/_h_env/Library/share/ament_cmake_copyright/cmake/ament_copyright.cmake:38 (message):
  ament_copyright() could not find program 'ament_copyright'
```

Which comes from
```
  find_program(ament_copyright_BIN NAMES "ament_copyright")
  if(NOT ament_copyright_BIN)
    message(FATAL_ERROR "ament_copyright() could not find program 'ament_copyright'")
  endif()
```

So we either have to figure out which package executables should not be installed in `lib/package-name`, or fix the cmake to also look in `lib/package-name`. I'm not sure whether ament_copyright is going to be the only package which is affected - I am guessing ament_linter and whatever else is out there might also be affected.

Any ideas @wolfv @traversaro?

wolfv, 2022-01-26:
Maybe we can also ping @ooeygui and @nuclearsandwich ... maybe they know how to identify the packages where the executable should end up in `lib` vs. `bin`?

wolfv, 2022-01-26:
Hmm, I am wondering, what happens if we don't give that command line argument at all?

traversaro, 2022-01-26:
We can also check what colcon or bloom are doing, at least in the `demo_nodes_py` it was doing the right thing with no additional options.

wolfv, 2022-01-26:
Yeah, I was just digging more into it ... some packages have a setup.cfg file:

https://github.com/ros2/demos/blob/master/demo_nodes_py/setup.cfg

That seems to give an argument for the `install_script` argument that we pass manually. It's also found in this test package: https://github.com/colcon/colcon-bundle/blob/84dd04d238fa2699b7c804e51733bad4b9aed29e/integration/ros2_workspace/src/test_py_package/setup.cfg

In ament_lint we don't find that setup.cfg file: https://github.com/ament/ament_lint/tree/master/ament_copyright

I didn't find anything in the ms-iot pipelines and I also didn't find anything more in colcon so far. They do some setup.py introspection but nothing that changes the `--install-scripts` arguments as far as I can see (https://github.com/colcon/colcon-python-setup-py/tree/master/colcon_python_setup_py)

So maybe not setting anything is the solution?

nuclearsandwich, 2022-01-26:
> Maybe we can also ping @ooeygui and @nuclearsandwich ... maybe they know how to identify the packages where the executable should end up in `lib` vs. `bin`?

In order to get the scripts into the /opt/ros/bin directory our bloom templates usually override the install_scripts dir for python packages. However when a package is designed to use the ros2run/libexec pattern they use the same mechanism to set the install_scripts to the libexec dir in [setup.cfg](https://github.com/ros2/demos/blob/master/demo_nodes_py/setup.cfg). Bloom is configured to look for this:

https://github.com/ros-infrastructure/bloom/blob/ab8ebb8ce2503801054f4a34137a3aec3a843ad7/bloom/generators/debian/generator.py#L395-L398

and only override the setting via pybuild if there's none configured in setup.cfg.
https://github.com/ros-infrastructure/bloom/blob/ab8ebb8ce2503801054f4a34137a3aec3a843ad7/bloom/generators/debian/templates/ament_python/rules.em#L24

Tobias-Fischer, 2022-01-26:
Ah, interesting, thanks for this @nuclearsandwich! The remaining code of interest: https://github.com/ros-infrastructure/bloom/blob/53beb5284fbe7b453af7cad3ef62d159168862bf/bloom/generators/debian/generator.py#L395-L406

So we need to check whether a setup.cfg exists and an install option is specified. A bit trickier with vinca, as we neither have access to the source files at this stage, and use a bash script instead of python. Any ideas @wolfv?

wolfv, 2022-01-27:
I guess we need to get our bash-foo going :)

On Linux / macOS we could try something like 

```
INSTALL_SCRIPTS_ARG=--install-scripts=$PREFIX/lib
if grep -q "install[-_]scripts" x.cfg; then
  INSTALL_SCRIPTS_ARG=""
fi
echo "ARG: $INSTALL_SCRIPTS_ARG"
```

On Windows I think we can use `findstr`. Going to figure that out now :)

wolfv, 2022-01-27:
On Windows, this should work:

```bat
set "INSTALL_SCRIPTS_ARG=--install-scripts=%_LIBRARY_PREFIX_PATH%\lib"
findstr install[-_]scripts setup.cfg
if %errorlevel% 0 (
  set INSTALL_SCRIPTS_ARG=""
)
echo %INSTALL_SCRIPTS_ARG%
```

Tobias-Fischer, 2022-02-02:
Needs https://github.com/RoboStack/vinca/pull/29 & #75

Tobias-Fischer, 2022-03-31:
Hopefully fixed with the latest rebuild, to be confirmed though

traversaro, 2022-03-31:
demo-nodes-py still needs to be rebuilt:
~~~
mamba create -n ros_galactic ros-galactic-demo-nodes-py python=3.9 -c robostack-experimental -c conda-forge --no-channel-priority --override-channels
Encountered problems while solving:
  - package ros-galactic-demo-nodes-py-0.14.3-py38hae75d3f_0 requires python_abi 3.8.*, but none of the providers can <..>
~~~

traversaro, 2022-04-01:
The executable is now correctly found, but another error emerged:
~~~
ros2 run demo_nodes_py demo_talker
(ros2) C:\Users\STraversaro>ros2 run demo_nodes_py demo_talker
Traceback (most recent call last):
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\Library\lib\demo_nodes_py\demo_talker-script.py", line 33, in <module>
    sys.exit(load_entry_point('demo-nodes-py==0.14.3', 'console_scripts', 'demo_talker')())
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\Library\lib\demo_nodes_py\demo_talker-script.py", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\lib\importlib\metadata.py", line 86, in load
    module = import_module(match.group('module'))
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\Lib\site-packages\demo_nodes_py\topics\talker.py", line 16, in <module>
    from rclpy.node import Node
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\Lib\site-packages\rclpy\node.py", line 43, in <module>
    from rclpy.client import Client
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\Lib\site-packages\rclpy\client.py", line 22, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\Lib\site-packages\rclpy\impl\implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\Lib\site-packages\rpyutils\import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'
The C extension 'C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\Lib\site-packages\_rclpy_pybind11.cp39-win_amd64.pyd' isn't present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#import-failing-without-library-present-on-the-system' for possible solutions
~~~

Environment created with:
~~~
mamba create -n ros2 -c conda-forge -c robostack-experimental ros-galactic-desktop
~~~

traversaro, 2022-04-01:
The problem can be reproduced more easily without any use of `ros2 run`:
~~~
(ros2) C:\Users\STraversaro>python -c "from rclpy.node import Node"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\Lib\site-packages\rclpy\node.py", line 43, in <module>
    from rclpy.client import Client
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\Lib\site-packages\rclpy\client.py", line 22, in <module>
    from rclpy.impl.implementation_singleton import rclpy_implementation as _rclpy
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\Lib\site-packages\rclpy\impl\implementation_singleton.py", line 32, in <module>
    rclpy_implementation = import_c_library('._rclpy_pybind11', package)
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\Lib\site-packages\rpyutils\import_c_library.py", line 39, in import_c_library
    return importlib.import_module(name, package=package)
  File "C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11'
The C extension 'C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2\Lib\site-packages\_rclpy_pybind11.cp39-win_amd64.pyd' isn't present on the system. Please refer to 'https://index.ros.org/doc/ros2/Troubleshooting/Installation-Troubleshooting/#import-failing-without-library-present-on-the-system' for possible solutions
~~~

So it could be considered a different issue.

traversaro, 2022-04-01:
With the workaround in https://github.com/RoboStack/ros-galactic/issues/96#issuecomment-1086014185, the node finally starts fine:
~~~
(ros2) C:\Users\STraversaro>ros2 run demo_nodes_py demo_talker
[INFO] [1648825676.436909000] [talker]: Publishing: "Hello World: 0"
[INFO] [1648825677.348159400] [talker]: Publishing: "Hello World: 1"
[INFO] [1648825678.347715300] [talker]: Publishing: "Hello World: 2"
[INFO] [1648825679.359195500] [talker]: Publishing: "Hello World: 3"
~~~
and so on so forth

Tobias-Fischer, 2022-04-03:
See #96 instead as this issue is resolved

closed 2022-04-03 by Tobias-Fischer

traversaro, 2022-04-03:
Finally working fine after fixing https://github.com/RoboStack/ros-galactic/issues/96 : 
~~~
(ros2) C:\Users\STraversaro>ros2 run demo_nodes_py demo_talker
[INFO] [1648985579.630156000] [talker]: Publishing: "Hello World: 0"
[INFO] [1648985580.556038200] [talker]: Publishing: "Hello World: 1"
[INFO] [1648985581.542916100] [talker]: Publishing: "Hello World: 2"
[INFO] [1648985582.546812300] [talker]: Publishing: "Hello World: 3"
~~~

<details>
  <summary>Full Environment</summary

~~~
(ros2) C:\Users\STraversaro>mamba list
# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2:
#
# Name                    Version                   Build  Channel
argcomplete               2.0.0              pyhd8ed1ab_0    conda-forge
assimp                    5.2.3                hc2aa0de_1    conda-forge
atomicwrites              1.4.0              pyh9f0ad1d_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
boost                     1.74.0           py39hefe7e4c_5    conda-forge
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
brotli                    1.0.9                h8ffe710_7    conda-forge
brotli-bin                1.0.9                h8ffe710_7    conda-forge
bullet                    3.21                 h832f523_1    conda-forge
bullet-cpp                3.21                 h2e25243_1    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
cairo                     1.16.0            h15b3021_1010    conda-forge
catkin_pkg                0.4.24             pyhd8ed1ab_1    conda-forge
certifi                   2021.10.8        py39hcbf5309_2    conda-forge
cffi                      1.15.0           py39h0878f49_0    conda-forge
cmake                     3.23.0               h39d44d4_1    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
cppcheck                  2.6.2            py39hd48a9a8_0    conda-forge
cryptography              36.0.2           py39h7bc7c5c_0    conda-forge
curl                      7.82.0               h789b8ee_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.18.1           py39hcbf5309_1    conda-forge
double-conversion         3.2.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.8                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flake8                    4.0.1              pyhd8ed1ab_2    conda-forge
flann                     1.9.1             h075982a_1010    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.96              hce3cb01_2    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.31.2           py39hb82d6ee_0    conda-forge
foonathan-memory          0.7.1                h57928b3_0    conda-forge
freeglut                  3.2.2                h0e60522_1    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
getopt-win32              0.1                  h8ffe710_0    conda-forge
gettext                   0.19.8.1          ha2e2712_1008    conda-forge
gl2ps                     1.4.2                h0597ee9_0    conda-forge
glew                      2.1.0                h39d44d4_2    conda-forge
gmock                     1.11.0               h2d74725_0    conda-forge
graphite2                 1.3.13                     1000    conda-forge
graphviz                  3.0.0                had6c3a3_1    conda-forge
gtest                     1.11.0               h2d74725_0    conda-forge
[... 409 lines cut ...]
vs2015_runtime            14.29.30037          h902a5da_6    conda-forge
vtk                       9.1.0           qt_py39h1ab545e_207    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
win32_setctime            1.1.0              pyhd8ed1ab_0    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.0.10               hcd874cb_0    conda-forge
xorg-libsm                1.2.3             hcd874cb_1000    conda-forge
xorg-libx11               1.7.2                hcd874cb_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                hcd874cb_1    conda-forge
xorg-libxpm               3.5.13               hcd874cb_0    conda-forge
[... cut ...]
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yaml-cpp                  0.6.3                ha925a31_4    conda-forge
zipp                      3.7.0              pyhd8ed1ab_1    conda-forge
zlib                      1.2.11            h8ffe710_1014    conda-forge
zstd                      1.5.2                h6255e5f_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
~~~

</details>

danzimmerman, 2022-06-23:
I'm having trouble with this with a fresh install as of yesterday, for example.

```
(galactic-clean) C:\Code\ros2\galactic_ws>ros2 run demo_nodes_py talker
No executable found
```

```
(galactic-clean) C:\Code\ros2\galactic_ws>ros2 pkg list | findstr demo_nodes
demo_nodes_cpp
demo_nodes_cpp_native
demo_nodes_py
```

Re: discussion in #96/#97, no problem there.

Is it possible this is related to `AMENT_PYTHON_EXECUTABLE` issues? (https://github.com/RoboStack/ros-galactic/issues/53#issuecomment-1164986782)

```
(galactic-clean) C:\Code\ros2\galactic_ws>dir %CONDA_PREFIX%\Lib\site-packages\rclpy | findstr _rclpy
04/03/2022  03:26 AM           497,664 _rclpy_pybind11.pyd
04/03/2022  03:26 AM            12,288 _rclpy_signal_handler.pyd
```

### Full Environment Info

`mamba info`

<details>

```
(galactic-clean) C:\Code\ros2\galactic_ws>mamba info

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (0.22.1) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

     active environment : galactic-clean
    active env location : C:\Users\dan\mambaforge\envs\galactic-clean
            shell level : 2
       user config file : C:\Users\dan\.condarc
 populated config files : C:\Users\dan\mambaforge\.condarc
                          C:\Users\dan\mambaforge\envs\galactic-clean\.condarc
          conda version : 4.12.0
    conda-build version : not installed
         python version : 3.9.13.final.0
       virtual packages : __cuda=11.6=0
                          __win=0=0
                          __archspec=1=x86_64
       base environment : C:\Users\dan\mambaforge  (writable)
      conda av data dir : C:\Users\dan\mambaforge\etc\conda
  conda av metadata url : None
[... 535 lines cut ...]
vtk                       9.1.0           qt_py39h1ab545e_208    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
win32_setctime            1.1.0              pyhd8ed1ab_0    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.0.10               hcd874cb_0    conda-forge
xorg-libsm                1.2.3             hcd874cb_1000    conda-forge
xorg-libx11               1.7.2                hcd874cb_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                hcd874cb_1    conda-forge
xorg-libxpm               3.5.13               hcd874cb_0    conda-forge
xorg-libxt                1.2.1                hcd874cb_2    conda-forge
xorg-xextproto            7.3.0             hcd874cb_1002    conda-forge
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yaml-cpp                  0.6.3                ha925a31_4    conda-forge
zipp                      3.8.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.12               h8ffe710_1    conda-forge
zstd                      1.5.2                h6255e5f_1    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
```
</details>

`C:\Users\dan\.condarc` does not actually exist on my system.

danzimmerman, 2022-06-23:
I can build and run my own Python node even with the `AMENT_PYTHON_EXECUTABLE` error 🤔 

```
(galactic-clean) C:\Code\ros2\galactic_ws>install\setup.bat
error: AMENT_PYTHON_EXECUTABLE 'C:\Users\dan\mambaforge\envs\galactic-clean\Library\python.exe' doesn't exist

(galactic-clean) C:\Code\ros2\galactic_ws>ros2 run pw_hello pw_hello_pynode
Hi from pw_hello.
```