RoboStack/ros-galactic #77: osx: turtlesim_node window doesn't come out

nikolausWest, 2022-01-28:
Having the same issue as in: https://github.com/RoboStack/ros-noetic/issues/158

I'm on macOS Monterey 12.0.1 and successfully installed ros-galactic using the [getting started instructions](https://robostack.github.io/GettingStarted.html) and when I run `ros2 run turtlesim turtlesim_node` the window doesn't come up.

Like the in the ros-noetic issue I get a terminal icon in the dock but the terminal I ran the run command in doesn't respond.

Tobias-Fischer, 2022-01-28:
Thanks for the report! Very annoying bug unfortunately - I'm not sure how to debug it. If anyone has ideas that would be appreciated :)

Tobias-Fischer, 2022-02-01:
So this can be fixed by using a newer version of qt. Will fix it in the next rebuild.

Eddie-717314-Huang, 2022-02-04:
hey, @nikolausWest have you tried this command line? https://github.com/RoboStack/ros-galactic/issues/73
I had exact same problem like you, but now I can run it. try it. cheers

nikolausWest, 2022-03-10:
Haven't tried that yet but will do!

Tobias-Fischer, 2022-04-03:
Should be fixed, but mind #88

closed 2022-04-03 by Tobias-Fischer