RoboStack/ros-galactic #78: zsh: command not found: roscore | Mac M1 | iTerm2

Eddie-717314-Huang, 2022-02-04:
I've followed the instructions on Robostack Getting Started steps by steps,
the only thing can't install at the moment is "colcon-common-extensions" which I've reported it on Gitter already, 
I can still use mamba to activate the ROS environment, and as I type "mamba list",  it showed all the ros-galactic-XXX packages, 
however, when I typed "roscore", the error showed "zsh: command not found: roscore"

I've tried to google the issue, didn't find the exact solutions, need some help here, thanks
this is where the path to setup.zsh:
/Users/eddie/mambaforge/envs/ROS/setup.zsh

I've tried to souce it to bash_profile and then vim the zshrc under the User Configuration, but it didn't work.

Tobias-Fischer, 2022-02-04:
There is no `roscore` in ROS2 (Galactic). Either install Noetic, or follow the Galactic tutorials.

closed 2022-02-04 by Tobias-Fischer

Eddie-717314-Huang, 2022-02-04:
I found an error install noetic, it showed "Encountered problems while solving:
  - package ros-noetic-desktop-1.5.0-py39hd384bd2_2 requires python 3.9.*, but none of the providers can be installed"
 but as I use "mamba install python=3.9.X" switch 3.8 to 3.9, reinstall the ros-noetic-desktop, it showed the opposite, it requires python 3.8?

Tobias-Fischer, 2022-02-04:
Please create a separate issue in the ros-noetic repo, with full information of what platform you are on, and all commands that you run.

Eddie-717314-Huang, 2022-02-04:
okay, thanks for the reminding, I'll close the issue, and make a new one.