RoboStack/ros-galactic #80: After running colcon build, the file install/setup.bash is not generated.

dave20874, 2022-02-11:
I've installed Robostack/ros-galactic now under Ubuntu 20.04 native, WSL (Ubuntu 20.04) and WSL2 (Ubuntu 20.04).  I started by installing mamba-forge and followed all the steps on the Getting Started page at robostack.github.io.  Each time, after completing a colcon build, I find there's no install/setup.bash file to establish the runtime environment.  There is an install/setup.sh file.  In testing this, I've noticed the packages colcon-argcomplete, colcon-bash, colcon-cd, and colcon-zsh are not installed along with colcon-common-extensions.  (These are all associated with the modifier 'sys_platform != 'win32'') This seems to be part of the problem but I'm no expert with conda/mamba, ROS 2 or colcon.

Tobias-Fischer, 2022-04-04:
Hi @dave20874 - sorry for taking a while to get back to you.

I am having difficulties understanding what exactly goes on. First of all, for ease, can we focus on the linux-64 platform (Ubuntu 20.04)? I understand you followed the instructions at robostack.github.io which is great.

Could you please outline the exact commands that you run following this? It would be great if you could tell us how to exactly replicate your problem so we can help you debugging. Which package do you try to build, how do you call colcon, ... ?

You also mention some colcon extension packages that are missing on conda-forge. We can certainly add them - is it only these four, or other ones, too?

Tobias-Fischer, 2022-04-04:
See https://github.com/conda-forge/staged-recipes/pull/18576 for the missing colcon extension packages - this will need to be merged, and then these packages will need to be added to colcon-extensions.

Tobias-Fischer, 2022-04-05:
You can now `mamba install colcon-argcomplete colcon-bash colcon-cd colcon-zsh` - does that help?

Tobias-Fischer, 2022-04-13:
Closing while waiting to hear back - I am assuming that installing `colcon-bash` fixes this issue. Feel free to reopen if this is not the case.

closed 2022-04-13 by Tobias-Fischer

dave20874, 2022-04-13:
Oops, my turn to apologize.  I've been busy with other issues and haven't tried your suggestions yet.  I'll try to retest soon and will let you know how it goes.

dave20874, 2022-04-14:
I've retested and, yes, adding colcon-bash fixed the issue.  Also, I removed and re-created my whole environment from scratch and with these fixes, everything worked perfectly.  Thank you.