RoboStack/ros-galactic #81: rviz2 does not start on OS X M1

KohlhardtC, 2022-02-22:
I followed the install instructions and confirmed that at least some ros2 commands worked as expected (e.g. ros2 topic list, ros2 run tf2_tools view_frames)

Here are the steps I followed to start Rviz:

`conda activate ros_galactic`
`rviz2`

**Expected result:** Rviz starts up

**Actual result**: a process named rviz2 does start and immediately allocates 100% CPU in Activity Monitor. An Rviz icon does show up in the dock, however no window ever shows up showing the actual Rviz interface. I waited at least 10 minutes.

An .rviz directory did appear in my home directory but it contained nothing

The ~/.ros/log directory did not contain anything helpful

**Environment:**
macOS Monterey
MacBook Pro Apple M1 Max

Tobias-Fischer, 2022-02-22:
Does https://github.com/RoboStack/ros-galactic/issues/73 help?

Tobias-Fischer, 2022-02-22:
Can you provide the output of `conda list`?

KohlhardtC, 2022-02-22:
```
(base) chrisk@ChrisKsMBP ros2-RoboStack % conda list
# packages in environment at /Users/chrisk/opt/miniconda3:
#
# Name                    Version                   Build  Channel
brotlipy                  0.7.0           py39h9ed2024_1003  
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
ca-certificates           2021.10.8            h033912b_0    conda-forge
certifi                   2021.10.8        py39h6e9494a_1    conda-forge
cffi                      1.15.0           py39hc55c11b_1  
charset-normalizer        2.0.4              pyhd3eb1b0_0  
conda                     4.11.0           py39h6e9494a_0    conda-forge
conda-content-trust       0.1.1              pyhd3eb1b0_0  
conda-package-handling    1.7.3            py39h9ed2024_1  
cryptography              36.0.0           py39hf6deb26_0  
icu                       68.2                 he49afe7_0    conda-forge
idna                      3.3                pyhd3eb1b0_0  
krb5                      1.19.2               hcfbf3a7_3    conda-forge
libarchive                3.5.2                h2b60450_1    conda-forge
libcurl                   7.80.0               h6dfd666_0  
libcxx                    12.0.0               h2f01273_0  
libedit                   3.1.20191231         h0678c8f_2    conda-forge
libev                     4.33                 haf1e3a3_1    conda-forge
libffi                    3.3                  hb1e8313_2  
libiconv                  1.16                 haf1e3a3_0    conda-forge
libnghttp2                1.46.0               ha29bfda_0  
libsolv                   0.7.19               hcf210ce_5    conda-forge
libssh2                   1.10.0               h52ee1ee_2    conda-forge
libxml2                   2.9.12               h93ec3fd_0    conda-forge
lz4-c                     1.9.3                he49afe7_1    conda-forge
lzo                       2.10              haf1e3a3_1000    conda-forge
mamba                     0.15.3           py39hb671511_0    conda-forge
ncurses                   6.3                  hca72f7f_2  
openssl                   1.1.1l               h0d85af4_0    conda-forge
pip                       21.2.4           py39hecd8cb5_0  
pycosat                   0.6.3            py39h9ed2024_0  
pycparser                 2.21               pyhd3eb1b0_0  
pyopenssl                 21.0.0             pyhd3eb1b0_1  
pysocks                   1.7.1            py39hecd8cb5_0  
python                    3.9.7                h88f2d9e_1  
python.app                3                py39hca72f7f_0  
python_abi                3.9                      2_cp39    conda-forge
readline                  8.1.2                hca72f7f_1  
reproc                    14.2.3               h0d85af4_0    conda-forge
reproc-cpp                14.2.3               he49afe7_0    conda-forge
requests                  2.27.1             pyhd3eb1b0_0  
ruamel_yaml               0.15.100         py39h9ed2024_0  
setuptools                58.0.4           py39hecd8cb5_0  
six                       1.16.0             pyhd3eb1b0_0  
sqlite                    3.37.0               h707629a_0  
tk                        8.6.11               h7bc2e8c_0  
tqdm                      4.62.3             pyhd3eb1b0_1  
tzdata                    2021e                hda174b7_0  
urllib3                   1.26.7             pyhd3eb1b0_0  
wheel                     0.37.1             pyhd3eb1b0_0  
xz                        5.2.5                h1de35cc_0  
yaml                      0.2.5                haf1e3a3_0  
zlib                      1.2.11               h4dc903c_4  
zstd                      1.5.0                hcb37349_1
```

Tobias-Fischer, 2022-02-22:
After `conda activate ros_galactic`; conda list of the base env isn't too important.

KohlhardtC, 2022-02-23:
I don't understand. Are you asking me to do something different?

Tobias-Fischer, 2022-02-23:
Yes. Please provide the output of

```
conda activate ros_galactic
conda list
```

KohlhardtC, 2022-02-23:
I see. I forgot to run _conda activate ros_galactic_ after I restarted. Sorry about that! Here you go. 

```
(base) chrisk@ChrisKsMBP ros2-RoboStack % conda activate ros_galactic
objc[9076]: Class AMSupportURLConnectionDelegate is implemented in both /usr/lib/libauthinstall.dylib (0x1dc18ab90) and /Library/Apple/System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/MobileDevice (0x106e042c8). One of the two will be used. Which one is undefined.
objc[9076]: Class AMSupportURLSession is implemented in both /usr/lib/libauthinstall.dylib (0x1dc18abe0) and /Library/Apple/System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/MobileDevice (0x106e04318). One of the two will be used. Which one is undefined.
objc[9077]: Class AMSupportURLConnectionDelegate is implemented in both /usr/lib/libauthinstall.dylib (0x1dc18ab90) and /Library/Apple/System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/MobileDevice (0x10a7bc2c8). One of the two will be used. Which one is undefined.
objc[9077]: Class AMSupportURLSession is implemented in both /usr/lib/libauthinstall.dylib (0x1dc18abe0) and /Library/Apple/System/Library/PrivateFrameworks/MobileDevice.framework/Versions/A/MobileDevice (0x10a7bc318). One of the two will be used. Which one is undefined.
(ros_galactic) chrisk@ChrisKsMBP ros2-RoboStack % conda list
# packages in environment at /Users/chrisk/opt/miniconda3/envs/ros_galactic:
#
# Name                    Version                   Build  Channel
aom                       3.2.0                he49afe7_2    conda-forge
argcomplete               2.0.0              pyhd8ed1ab_0    conda-forge
assimp                    5.1.6                h1224e73_0    conda-forge
atk-1.0                   2.36.0               he69c4ee_4    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
boost                     1.74.0           py38hb0f0857_5    conda-forge
boost-cpp                 1.74.0               hff03dee_4    conda-forge
brotli                    1.0.9                h0d85af4_6    conda-forge
brotli-bin                1.0.9                h0d85af4_6    conda-forge
bullet-cpp                3.21                 h06619e7_1    conda-forge
bzip2                     1.0.8                h0d85af4_4    conda-forge
c-ares                    1.18.1               h0d85af4_0    conda-forge
c-compiler                1.4.0                h774085c_0    conda-forge
ca-certificates           2021.10.8            h033912b_0    conda-forge
cairo                     1.16.0            he43a7df_1008    conda-forge
catkin_pkg                0.4.24             pyhd8ed1ab_1    conda-forge
cctools                   973.0.1              hd9211c8_6    conda-forge
cctools_osx-64            973.0.1              h609b070_6    conda-forge
certifi                   2021.10.8        py38h50d1736_1    conda-forge
cffi                      1.15.0           py38h1a44b6c_0    conda-forge
clang                     12.0.1               h694c41f_4    conda-forge
clang-12                  12.0.1          default_he082bbe_4    conda-forge
clang_osx-64              12.0.1               h633439f_6    conda-forge
clangxx                   12.0.1          default_he082bbe_4    conda-forge
clangxx_osx-64            12.0.1               hdb584c0_6    conda-forge
cmake                     3.22.2               h4d639be_0    conda-forge
colcon-cmake              0.2.26                     py_0    conda-forge
colcon-common-extensions  0.2.1            py38h50d1736_4    conda-forge
colcon-core               0.7.1              pyhd8ed1ab_0    conda-forge
colcon-defaults           0.2.5                      py_0    conda-forge
colcon-devtools           0.2.2                      py_0    conda-forge
colcon-library-path       0.2.1                      py_0    conda-forge
colcon-metadata           0.2.5                      py_0    conda-forge
colcon-output             0.2.12                     py_0    conda-forge
colcon-package-information 0.3.3                      py_0    conda-forge
colcon-package-selection  0.2.10                     py_0    conda-forge
colcon-parallel-executor  0.2.4                      py_0    conda-forge
colcon-pkg-config         0.1.0                      py_0    conda-forge
colcon-powershell         0.3.6                      py_0    conda-forge
colcon-python-setup-py    0.2.7                      py_0    conda-forge
colcon-recursive-crawl    0.2.1                      py_0    conda-forge
colcon-ros                0.3.21                     py_0    conda-forge
colcon-test-result        0.3.8                      py_0    conda-forge
coloredlogs               15.0.1           py38h50d1736_2    conda-forge
compiler-rt               12.0.1               he01351e_0    conda-forge
compiler-rt_osx-64        12.0.1               hd3f61c9_0    conda-forge
compilers                 1.4.0                h694c41f_0    conda-forge
console_bridge            1.0.1                h65a07b1_0    conda-forge
coverage                  6.3.1            py38h96a0964_0    conda-forge
cppcheck                  2.6.2            py38h0a3ca99_0    conda-forge
cryptography              36.0.1           py38h56c4533_0    conda-forge
curl                      7.81.0               hf45b732_0    conda-forge
cxx-compiler              1.4.0                hbb4e6a2_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
[... cut ...]
xz                        5.2.5                haf1e3a3_1    conda-forge
yaml                      0.2.5                h0d85af4_2    conda-forge
yaml-cpp                  0.6.3                hb1e8313_4    conda-forge
zipp                      3.7.0              pyhd8ed1ab_1    conda-forge
zlib                      1.2.11            h9173be1_1013    conda-forge
zstd                      1.5.2                h582d3a0_0    conda-forge
zziplib                   0.13.69              h5dbffcc_1    conda-forge
```

Tobias-Fischer, 2022-02-23:
It seems that you use the osx-64 packages. Can you use the osx-arm64 ones instead?

KohlhardtC, 2022-02-23:
Sorry I'm new to all this and don't understand how best to use the osx-arm64 packages. Can you please point me to directions for making the switch?

Tobias-Fischer, 2022-02-24:
So it seems like you installed a miniconda3 that uses the Intel (osx-64) packages by default. Seeing that you have an M1, it would be better for you to use the M1 Apple Silicon (osx-arm64) packages.

To do that, I would recommend starting with a fresh installation of conda, using the Apple Silicon version of Mambaforge on https://github.com/conda-forge/miniforge (i.e. https://github.com/conda-forge/miniforge/releases/latest/download/Mambaforge-MacOSX-arm64.sh).

Then follow the installation instructions again, and it will by default install the osx-arm64 packages. With these packages, I can run rviz2 just fine on my M1 MacBook (after following https://github.com/RoboStack/ros-galactic/issues/73).

Does that make sense?

KohlhardtC, 2022-02-24:
I'm happy to report that this solve my issue. Thank you!