RoboStack/ros-galactic #82: Windows ros2cli not found

AndreasGronlund, 2022-03-22:
ros-galactic installed on Windows 10 using miniconda and mamba. Following the installation instructions and starting rviz2 gives

```
(ros_galactic) PS C:\dev> ros2 run rviz2 rviz2
Traceback (most recent call last):
  File "C:\dev\Miniconda3\envs\ros_galactic\Library\bin\ros2-script.py", line 33, in <module>
    sys.exit(load_entry_point('ros2cli==0.13.1', 'console_scripts', 'ros2')())
  File "C:\dev\Miniconda3\envs\ros_galactic\Library\bin\ros2-script.py", line 22, in importlib_load_entry_point
    for entry_point in distribution(dist_name).entry_points
  File "C:\dev\Miniconda3\envs\ros_galactic\lib\importlib\metadata.py", line 503, in distribution
    return Distribution.from_name(distribution_name)
  File "C:\dev\Miniconda3\envs\ros_galactic\lib\importlib\metadata.py", line 177, in from_name
    raise PackageNotFoundError(name)
importlib.metadata.PackageNotFoundError: ros2cli
```

I get the same error message when trying to start my own ros nodes, e.g. simple subscriber/talker. Following the same installation procedure in Ubuntu works fine.

Tobias-Fischer, 2022-03-22:
Hi,
Thanks for your interest in RoboStack! Unfortunately our windows packages are currently broken, see https://github.com/RoboStack/ros-galactic/issues/74 and let's keep track of the problem there.

I'm busy with work related things those days and I'm not sure yet when I'll get around to rebuild the packages, apologies for this!

closed 2022-03-22 by Tobias-Fischer

traversaro, 2022-04-03:
@AndreasGronlund fyi after fixing https://github.com/RoboStack/ros-galactic/issues/74 `ros2 run rviz2 rviz2` works fine:

[image]