RoboStack/ros-galactic #91: Create activate scripts for PowerShell

doriantsai, 2022-03-29:
I am trying to get Robostack to work in with Windows 10 Powershell and running into an error. Will post it shortly.

Tobias-Fischer, 2022-03-29:
See also https://github.com/RoboStack/ros-noetic/issues/21

@traversaro - you mentioned that it should be easy-ish to create the ps1 script in vinca. Do you have any documentation that we could look at?

traversaro, 2022-03-29:
> @traversaro - you mentioned that it should be easy-ish to create the ps1 script in vinca. Do you have any documentation that we could look at?

Not sure, but apparently both colcon and ros2 docs mention `.ps1` activation scripts:
* https://colcon.readthedocs.io/en/released/developer/environment.html
* https://docs.ros.org/en/galactic/Tutorials/Configuring-ROS2-Environment.html

That is the reason why I was more optimistic of supporting PowerShell on ROS2 then on ROS1, as mentioned in https://github.com/RoboStack/ros-noetic/issues/21 .

traversaro, 2022-03-29:
For the vinca part, probably we will just modify around this line: https://github.com/RoboStack/vinca/blob/2b88f42fdb11e98dbcabda0b6a828a57d2d2a6df/vinca/template.py#L89 and add the approriate templates in https://github.com/RoboStack/vinca/tree/2b88f42fdb11e98dbcabda0b6a828a57d2d2a6df/vinca/templates .

closed 2023-01-25 by Tobias-Fischer