RoboStack/ros-galactic #96: ModuleNotFoundError: No module named 'rclpy._rclpy_pybind11' on Windows on Python 3.9

traversaro, 2022-04-01:


traversaro, 2022-04-01:
A workaround is renaming `%CONDA_PREFIX%\Lib\site-packages\rclpy\_rclpy_pybind11.dll` and `%CONDA_PREFIX%\Lib\site-packages\rclpy\_rclpy_signal_handler.dll` to  `%CONDA_PREFIX%\Lib\site-packages\rclpy\_rclpy_pybind11.pyd` and `%CONDA_PREFIX%\Lib\site-packages\rclpy\_rclpy_signal_handler.pyd`.

traversaro, 2022-04-01:
Related FAQ: https://docs.python.org/3/faq/windows.html#is-a-pyd-file-the-same-as-a-dll .

traversaro, 2022-04-01:
Out of the packages installed by ros-galactic-desktop, the two packages affected by this are `rclpy` and `rosbag2_py`. There are a lot of other packages that instead have correctly named `.pyd` files, but most of them are packages of msgs that have `<..>__rosidl_typesupport_c.pyd`, `<..>__rosidl_typesupport_fastrtps_c.pyd` and `<..>__rosidl_typesupport_introspection_c.pyd` files.

Tobias-Fischer, 2022-04-01:
Fixing osx-arm64 seems to have led to this error. I think all it takes is setting the IMPORT_SUFFIX as in https://cmake.cmake.narkive.com/azPoFCCq/can-i-change-the-extension-of-a-dll in the patches (see https://github.com/RoboStack/ros-galactic/blob/main/patch/ros-galactic-rclpy.patch for rclypy)

traversaro, 2022-04-01:
I think `IMPORT_SUFFIX` is to specify the extention of the `.lib` libraries. The code called by `pybind11_add_module` is https://github.com/pybind/pybind11/blob/b3ebd11d9851007f525e2e41de53b7913b24787c/tools/pybind11Tools.cmake#L132 : 
~~~
  # The prefix and extension are provided by FindPythonLibsNew.cmake
  set_target_properties(${name} PROPERTIES PREFIX "${PYTHON_MODULE_PREFIX}"
                                           SUFFIX "${PYTHON_MODULE_EXTENSION}")
~~~

traversaro, 2022-04-01:
I tried to provid a fix in https://github.com/RoboStack/ros-galactic/issues/96 , we can also add a `if(MSVC)` to be more sure of avoiding regressions.

closed 2022-04-03 by Tobias-Fischer

traversaro, 2022-04-03:
Working fine after https://github.com/RoboStack/ros-galactic/pull/97 : 

~~~
(ros2) C:\Users\STraversaro>ros2 run demo_nodes_py demo_talker
[INFO] [1648985579.630156000] [talker]: Publishing: "Hello World: 0"
[INFO] [1648985580.556038200] [talker]: Publishing: "Hello World: 1"
[INFO] [1648985581.542916100] [talker]: Publishing: "Hello World: 2"
[INFO] [1648985582.546812300] [talker]: Publishing: "Hello World: 3"
~~~

<details>
  <summary>Full Environment</summary

~~~
(ros2) C:\Users\STraversaro>mamba list
# packages in environment at C:\Users\STraversaro\AppData\Local\mambaforge\envs\ros2:
#
# Name                    Version                   Build  Channel
argcomplete               2.0.0              pyhd8ed1ab_0    conda-forge
assimp                    5.2.3                hc2aa0de_1    conda-forge
atomicwrites              1.4.0              pyh9f0ad1d_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
boost                     1.74.0           py39hefe7e4c_5    conda-forge
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
brotli                    1.0.9                h8ffe710_7    conda-forge
brotli-bin                1.0.9                h8ffe710_7    conda-forge
bullet                    3.21                 h832f523_1    conda-forge
bullet-cpp                3.21                 h2e25243_1    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
ca-certificates           2021.10.8            h5b45459_0    conda-forge
cairo                     1.16.0            h15b3021_1010    conda-forge
catkin_pkg                0.4.24             pyhd8ed1ab_1    conda-forge
certifi                   2021.10.8        py39hcbf5309_2    conda-forge
cffi                      1.15.0           py39h0878f49_0    conda-forge
cmake                     3.23.0               h39d44d4_1    conda-forge
colorama                  0.4.4              pyh9f0ad1d_0    conda-forge
console_bridge            1.0.2                h5362a0b_1    conda-forge
cppcheck                  2.6.2            py39hd48a9a8_0    conda-forge
cryptography              36.0.2           py39h7bc7c5c_0    conda-forge
curl                      7.82.0               h789b8ee_0    conda-forge
cycler                    0.11.0             pyhd8ed1ab_0    conda-forge
distro                    1.6.0              pyhd8ed1ab_0    conda-forge
docutils                  0.18.1           py39hcbf5309_1    conda-forge
double-conversion         3.2.0                h0e60522_0    conda-forge
eigen                     3.4.0                h2d74725_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.8                h39d44d4_0    conda-forge
ffmpeg                    4.3.1                ha925a31_0    conda-forge
flake8                    4.0.1              pyhd8ed1ab_2    conda-forge
flann                     1.9.1             h075982a_1010    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
font-ttf-source-code-pro  2.038                h77eed37_0    conda-forge
font-ttf-ubuntu           0.83                 hab24e00_0    conda-forge
fontconfig                2.13.96              hce3cb01_2    conda-forge
fonts-conda-ecosystem     1                             0    conda-forge
fonts-conda-forge         1                             0    conda-forge
fonttools                 4.31.2           py39hb82d6ee_0    conda-forge
foonathan-memory          0.7.1                h57928b3_0    conda-forge
freeglut                  3.2.2                h0e60522_1    conda-forge
freeimage                 3.18.0               hfcb8cd7_7    conda-forge
freetype                  2.10.4               h546665d_1    conda-forge
fribidi                   1.0.10               h8d14728_0    conda-forge
getopt-win32              0.1                  h8ffe710_0    conda-forge
gettext                   0.19.8.1          ha2e2712_1008    conda-forge
gl2ps                     1.4.2                h0597ee9_0    conda-forge
glew                      2.1.0                h39d44d4_2    conda-forge
gmock                     1.11.0               h2d74725_0    conda-forge
graphite2                 1.3.13                     1000    conda-forge
graphviz                  3.0.0                had6c3a3_1    conda-forge
[... 410 lines cut ...]
vs2015_runtime            14.29.30037          h902a5da_6    conda-forge
vtk                       9.1.0           qt_py39h1ab545e_207    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
win32_setctime            1.1.0              pyhd8ed1ab_0    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.0.10               hcd874cb_0    conda-forge
xorg-libsm                1.2.3             hcd874cb_1000    conda-forge
xorg-libx11               1.7.2                hcd874cb_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                hcd874cb_1    conda-forge
xorg-libxpm               3.5.13               hcd874cb_0    conda-forge
xorg-libxt                1.2.1                hcd874cb_2    conda-forge
[... cut ...]
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yaml-cpp                  0.6.3                ha925a31_4    conda-forge
zipp                      3.7.0              pyhd8ed1ab_1    conda-forge
zlib                      1.2.11            h8ffe710_1014    conda-forge
zstd                      1.5.2                h6255e5f_0    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge
~~~

</details>