RoboStack/ros-galactic #98: Fix build Pilz industrial motion planner on Win

Tobias-Fischer, 2022-04-03:
It complains that it's an ambiguous function call somewhere in tf2. For now I disabled the build of just this package.

Tobias-Fischer, 2022-04-03:
Error log:
```
2022-04-03T00:58:44.8919990Z   Building Custom Rule C:/bld/ros-galactic-pilz-industrial-motion-planner_1648947368018/work/ros-galactic-pilz-industrial-motion-planner/src/work/CMakeLists.txt
2022-04-03T00:58:44.8921113Z   trajectory_functions.cpp
2022-04-03T00:58:49.1149559Z %PREFIX%\Library\include\geometric_shapes/bodies.h(34,1): warning C4005: '_USE_MATH_DEFINES': macro redefinition [%SRC_DIR%\build\trajectory_generation_common.vcxproj]
2022-04-03T00:58:49.1515694Z %SRC_DIR%\ros-galactic-pilz-industrial-motion-planner\src\work\src\trajectory_functions.cpp : message : see previous definition of '_USE_MATH_DEFINES' [%SRC_DIR%\build\trajectory_generation_common.vcxproj]
2022-04-03T00:58:52.7594706Z %PREFIX%\Library\include\tf2/impl/convert.h(60,1): error C2668: 'tf2::fromMsg': ambiguous call to overloaded function [%SRC_DIR%\build\trajectory_generation_common.vcxproj]
2022-04-03T00:58:52.7597121Z %PREFIX%\Library\include\tf2_eigen/tf2_eigen.h(614,6): message : could be 'void Eigen::fromMsg(const geometry_msgs::msg::Pose &,Eigen::Isometry3d &)' [found using argument-dependent lookup] [%SRC_DIR%\build\trajectory_generation_common.vcxproj]
2022-04-03T00:58:52.7599327Z %PREFIX%\Library\include\tf2_eigen/tf2_eigen.h(437,6): message : or       'void tf2::fromMsg(const geometry_msgs::msg::Pose &,Eigen::Isometry3d &)' [%SRC_DIR%\build\trajectory_generation_common.vcxproj]
2022-04-03T00:58:52.7601399Z %PREFIX%\Library\include\tf2/convert.h(138,6): message : or       'void tf2::fromMsg<geometry_msgs::msg::Pose_<std::allocator<void>>,Eigen::Isometry3d>(const A &,B &)' [%SRC_DIR%\build\trajectory_generation_common.vcxproj]
2022-04-03T00:58:52.7602417Z           with
2022-04-03T00:58:52.7602915Z           [
2022-04-03T00:58:52.7603553Z               A=geometry_msgs::msg::Pose_<std::allocator<void>>,
2022-04-03T00:58:52.7604245Z               B=Eigen::Isometry3d
2022-04-03T00:58:52.7604789Z           ]
2022-04-03T00:58:52.7605646Z %PREFIX%\Library\include\tf2/impl/convert.h(60,1): message : while trying to match the argument list '(const A, B)' [%SRC_DIR%\build\trajectory_generation_common.vcxproj]
2022-04-03T00:58:52.7606511Z           with
2022-04-03T00:58:52.7607023Z           [
2022-04-03T00:58:52.7607581Z               A=geometry_msgs::msg::Pose
2022-04-03T00:58:52.7608159Z           ]
2022-04-03T00:58:52.7608661Z           and
2022-04-03T00:58:52.7609139Z           [
2022-04-03T00:58:52.7609735Z               B=Eigen::Isometry3d
2022-04-03T00:58:52.7610298Z           ]
2022-04-03T00:58:52.7611343Z %PREFIX%\Library\include\tf2/convert.h(151): message : see reference to function template instantiation 'void tf2::impl::Converter<true,false>::convert<A,B>(const A &,B &)' being compiled [%SRC_DIR%\build\trajectory_generation_common.vcxproj]
2022-04-03T00:58:52.7612399Z           with
2022-04-03T00:58:52.7613015Z           [
2022-04-03T00:58:52.7613579Z               A=geometry_msgs::msg::Pose,
2022-04-03T00:58:52.7614284Z               B=Eigen::Isometry3d
2022-04-03T00:58:52.7614798Z           ]
2022-04-03T00:58:52.7615778Z %PREFIX%\Library\include\tf2/convert.h(151): message : see reference to function template instantiation 'void tf2::impl::Converter<true,false>::convert<A,B>(const A &,B &)' being compiled [%SRC_DIR%\build\trajectory_generation_common.vcxproj]
2022-04-03T00:58:52.7616761Z           with
2022-04-03T00:58:52.7617208Z           [
2022-04-03T00:58:52.7617753Z               A=geometry_msgs::msg::Pose,
2022-04-03T00:58:52.7618569Z               B=Eigen::Isometry3d
2022-04-03T00:58:52.7619100Z           ]
2022-04-03T00:58:52.7620352Z %SRC_DIR%\ros-galactic-pilz-industrial-motion-planner\src\work\src\trajectory_functions.cpp(118): message : see reference to function template instantiation 'void tf2::convert<geometry_msgs::msg::Pose,Eigen::Isometry3d>(const A &,B &)' being compiled [%SRC_DIR%\build\trajectory_generation_common.vcxproj]
2022-04-03T00:58:52.7621619Z           with
2022-04-03T00:58:52.7622136Z           [
2022-04-03T00:58:52.7622720Z               A=geometry_msgs::msg::Pose,
2022-04-03T00:58:52.7623331Z               B=Eigen::Isometry3d
2022-04-03T00:58:52.7623889Z           ]
2022-04-03T00:58:53.0638693Z   trajectory_generator.cpp
```

traversaro, 2025-01-09:
Related PRs:
* https://github.com/ros/geometry2/pull/367
* https://github.com/ros/geometry2/pull/444
* https://github.com/ros/geometry2/pull/367

traversaro, 2025-01-09:
Upstream fix in moveit: https://github.com/moveit/moveit2/pull/3219 .

traversaro, 2025-01-09:
We do not mantain anymore galactic, but a fix for humble is suggested in https://github.com/RoboStack/ros-humble/pull/241 .