RoboStack/ros-galactic #103: MoveIt errors with `srdfdom.dll` missing from `%CONDA_PREFIX%\Library\bin` folder.

danzimmerman, 2022-06-24:
I couldn't add the MoveIt MotionPlanning plugin on a fresh Galactic install until I copied `srdfdom.dll` from `\Library\lib\srdfdom\` to `Library\bin`, seems like the `srdfdom` issue discussed in #36 I see https://github.com/ros-planning/srdfdom/pull/98 but that should have been picked up by now I suppose.

I'm up-to-date there:
```
(galactic-clean) C:\Code\ros2\galactic_ws>mamba list | findstr srdfdom
ros-galactic-srdfdom      2.0.3            py39h2545d16_3    robostack-experimental
```

Removing the `srdfdom.dll` in `Library\bin`  I get the following when I try to add the MotionPlanning plugin to Rviz2 

```
(galactic-clean) C:\Code\ros2\galactic_ws>del %CONDA_PREFIX%\library\bin\srdfdom.dll

(galactic-clean) C:\Code\ros2\galactic_ws>rviz2
[INFO] [1656050984.168199100] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [1656050984.168471900] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [1656050984.262447500] [rviz2]: Stereo is NOT SUPPORTED
[ERROR] [1656050998.524384100] [rviz2]: PluginlibFactory: The plugin for class 'moveit_rviz_plugin/MotionPlanning' failed to load. Error: Failed to load library C:\Users\dan\mambaforge\envs\galactic-clean\Library\bin\moveit_motion_planning_rviz_plugin.dll. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library LoadLibrary error: 126, at C:\bld\ros-galactic-rcutils_1648674430149\work\ros-galactic-rcutils\src\work\src\shared_library.c:159
```

I discovered this by trying to get the `ur_moveit_config` package from https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver working on Windows. When I launch that MoveIt config,  `move_group.exe` actually pops up an informative window which can be reproduced with:

```
(galactic-clean) C:\Code\ros2\galactic_ws>ros2 run moveit_ros_move_group move_group
```

[image]

With the DLL: 

```
(galactic-clean) C:\Code\ros2\galactic_ws>copy %CONDA_PREFIX%\Library\lib\srdfdom\srdfdom.dll %CONDA_PREFIX%\Library\bin\
        1 file(s) copied.

(galactic-clean) C:\Code\ros2\galactic_ws>ros2 run moveit_ros_move_group move_group
[WARN] [1656051436.859884700] [move_group.move_group]: MoveGroup launched without ~default_planning_pipeline specifying the namespace for the default planning pipeline configuration
[WARN] [1656051436.860105000] [move_group.move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
```

### Environment

`mamba list`

<details>

```
(galactic-clean) C:\Code\ros2\galactic_ws>mamba list
# packages in environment at C:\Users\<username>\mambaforge\envs\galactic-clean:
#
# Name                    Version                   Build  Channel
argcomplete               2.0.0              pyhd8ed1ab_0    conda-forge
assimp                    5.2.3                hc2aa0de_1    conda-forge
atomicwrites              1.4.0              pyh9f0ad1d_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
boost                     1.74.0           py39hefe7e4c_5    conda-forge
boost-cpp                 1.74.0               h9f4b32c_8    conda-forge
brotli                    1.0.9                h8ffe710_7    conda-forge
brotli-bin                1.0.9                h8ffe710_7    conda-forge
bullet                    3.21                 h832f523_2    conda-forge
bullet-cpp                3.21                 h2e25243_2    conda-forge
bzip2                     1.0.8                h8ffe710_4    conda-forge
c-compiler                1.4.2                h8ffe710_0    conda-forge
ca-certificates           2022.6.15            h5b45459_0    conda-forge
cairo                     1.16.0            h15b3021_1010    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
certifi                   2022.6.15        py39hcbf5309_0    conda-forge
cffi                      1.15.0           py39h0878f49_0    conda-forge
clangdev                  5.0.0                   flang_3    conda-forge
cmake                     3.23.2               h39d44d4_0    conda-forge
colcon-cmake              0.2.26                     py_0    conda-forge
[... 517 lines cut ...]
vswhere                   3.0.3                h57928b3_0    conda-forge
vtk                       9.1.0           qt_py39h1ab545e_208    conda-forge
wheel                     0.37.1             pyhd8ed1ab_0    conda-forge
win32_setctime            1.1.0              pyhd8ed1ab_0    conda-forge
xorg-kbproto              1.0.7             hcd874cb_1002    conda-forge
xorg-libice               1.0.10               hcd874cb_0    conda-forge
xorg-libsm                1.2.3             hcd874cb_1000    conda-forge
xorg-libx11               1.7.2                hcd874cb_0    conda-forge
xorg-libxau               1.0.9                hcd874cb_0    conda-forge
xorg-libxdmcp             1.1.3                hcd874cb_0    conda-forge
xorg-libxext              1.3.4                hcd874cb_1    conda-forge
xorg-libxpm               3.5.13               hcd874cb_0    conda-forge
xorg-libxt                1.2.1                hcd874cb_2    conda-forge
xorg-xextproto            7.3.0             hcd874cb_1002    conda-forge
xorg-xproto               7.0.31            hcd874cb_1007    conda-forge
xz                        5.2.5                h62dcd97_1    conda-forge
yaml                      0.2.5                h8ffe710_2    conda-forge
yaml-cpp                  0.6.3                ha925a31_4    conda-forge
zipp                      3.8.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.12               h8ffe710_1    conda-forge
zstd                      1.5.2                h6255e5f_1    conda-forge
zziplib                   0.13.69              h1d00b33_1    conda-forge

```
</details>

wolfv, 2022-06-24:
Hey, thanks for the nice writeup!

I was wondering if you could try with Humble? We've gotten pretty far with the packages on Windows, there, too (almost at `desktop-full`! :)

`mamba create -n moveithumble ros-humble-moveit ros-humble-desktop -c robostack-humble -c conda-forge`

We can probably rebuild a new version of `ros-galactic-srdfdom` (assuming they've released this by now). If you want to, you could just make a PR that changes the vinca_win.yaml and mention `srdfdom` to see if a new version pops up.

wolfv, 2022-06-24:
The version in galactic is 2.0.3, and the good version is 2.0.4.

https://github.com/ros/rosdistro/blob/2d3601dbc16e5c8c351ac60c1a48e2a67f10d03d/galactic/distribution.yaml#L5265-L5279

Tobias-Fischer, 2022-06-24:
@danzimmerman - if you add the patch from https://github.com/ros-planning/srdfdom/pull/98 and open a PR here to rebuild srdfdom I'd be happy to merge, if you need to stick to galactic.

danzimmerman, 2022-06-24:
@wolfv This definitely is fixed in Humble... I'm having this issue https://github.com/ros2/ros2cli/issues/637... are you seeing that?

@Tobias-Fischer Patching this for `galactic` would be a good first patch... will try to work on it when I get a chance.

traversaro, 2022-06-24:
> @wolfv This definitely is fixed in Humble... I'm now having this issue [ros2/ros2cli#637](https://github.com/ros2/ros2cli/issues/637)... are you seeing that?

Never saw that, but looking at the references there was a fix in https://github.com/ros2/ros2cli/pull/686 that is in Foxy but not in Galactic: https://github.com/ros2/ros2cli/blob/galactic/ros2cli/ros2cli/node/daemon.py#L104 .

danzimmerman, 2022-06-24:
Yeah something's broken with the daemon or my network/firewall config. The daemon is now using `daemon.get_xmlrpc_server_url()` instead of the Foxy fix
```
class DaemonNode:

    def __init__(self, args):
        self._args = args
        self._proxy = ServerProxy(
            daemon.get_xmlrpc_server_url(),
            allow_none=True)
        print(f'**  DaemonNode._proxy was set to {daemon.get_xmlrpc_server_url()} **')
        self._methods = []

    @property
    def connected(self):
        try:
            self._proxy.system.listMethods()
        except ConnectionRefusedError:
            print('*** Connection refused! ***')
            return False
        return True
```

It does seem to be refusing the connection.
```
(humble) C:\Code\ros2\galactic_ws>

(humble) C:\Code\ros2\galactic_ws>ros2 daemon start
The daemon is already running

(humble) C:\Code\ros2\galactic_ws>ros2 daemon status
**  DaemonNode._proxy was set to http://127.0.0.1:11528/ros2cli/ **
*** Connection refused! ***
The daemon is not running

(humble) C:\Code\ros2\galactic_ws>ros2 daemon start
The daemon is already running

(humble) C:\Code\ros2\galactic_ws>ros2 daemon stop
**  DaemonNode._proxy was set to http://127.0.0.1:11528/ros2cli/ **
*** Connection refused! ***
The daemon is not running
```

I'll dig in to that better later.

danzimmerman, 2022-06-24:
Humble also still has this `xacro` problem:

https://answers.ros.org/question/354579/ros-2-xacro-executable-will-not-run/#402816
https://github.com/ros/xacro/pull/304

I'll read up on the vinca/patching workflow and see what I can do when I get a chance