RoboStack/ros-galactic #104: Ignition Gazebo crashes on start

BenjaminNavarro, 2022-06-24:
I just created a new environement and installed `ros-galactic-ros-ign-gazebo`.

The installation went fine but running `ign gazebo` generate some errors in the console and the application crashes a second later.
```
> ign gazebo             
Error while loading the library [/home/benjamin/.conda/envs/ros2_env/lib/ign-physics-4/engine-plugins/libignition-physics-dartsim-plugin.so]: libode.so.0.16.2: cannot open shared object file: No such file or directory
[Err] [Physics.cc:600] Unable to load the [/home/benjamin/.conda/envs/ros2_env/lib/ign-physics-4/engine-plugins/libignition-physics-dartsim-plugin.so] library.
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin72139]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin98835]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin63996]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin59288]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin8327]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin20386]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin303]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin85117]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin78014]
[GUI] [Err] [RenderEngineManager.cc:461] Failed to load plugin [ogre2] : couldn't find shared library.
[GUI] [Err] [RenderUtil.cc:1678] Engine [ogre2] is not supported. Loading OGRE2 instead.
[GUI] [Err] [RenderEngineManager.cc:461] Failed to load plugin [ogre2] : couldn't find shared library.
Stack trace (most recent call last) in thread 255529:
#14   Object "", at 0xffffffffffffffff, in 
#13   Source "../sysdeps/unix/sysv/linux/x86_64/clone.S", line 100, in __clone [0x7f216cd11873]
#12   Source "/usr/src/debug/glibc/nptl/pthread_create.c", line 442, in start_thread [0x7f216cc8c54c]
#11   Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f2167aa190e, in 
#10   Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f2167aa094c, in QThread::exec()
#9    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f2167ba899a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#8    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f2167bd8bc0, in QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#7    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f2167bac1bf, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#6    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f2167babf87, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#5    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Widgets.so.5.12.9", at 0x7f216757a34c, in QApplication::notify(QObject*, QEvent*)
#4    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Widgets.so.5.12.9", at 0x7f2167574821, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#3    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f2167bc0f15, in QObject::event(QEvent*)
#2    Object "/home/benjamin/.conda/envs/ros2_env/lib/ign-gazebo-5/plugins/gui/libGzScene3D.so", at 0x7f214c4dd150, in ignition::gazebo::v5::RenderThread::RenderNext(ignition::gazebo::v5::RenderSync*)
#1    Object "/home/benjamin/.conda/envs/ros2_env/lib/ign-gazebo-5/plugins/gui/libGzScene3D.so", at 0x7f214c4d986f, in ignition::gazebo::v5::IgnRenderer::Initialize()
#0    Object "/home/benjamin/.conda/envs/ros2_env/lib/libignition-gazebo5-rendering.so.5.1.0", at 0x7f214c407aff, in ignition::gazebo::v5::RenderUtil::Init()
Segmentation fault (Address not mapped to object [(nil)])
```
It seems that it failed to locate the ODE and Ogre 2 shared libs. I don't know if they are not installed or if its just the `LD_LIBRARY_PATH` that is not set correctly in order for the dynamic linker to find them.

Tobias-Fischer, 2022-06-24:
What happens if you `mamba install libode=0.16.2`? After doing that, if it still doesn't work, please provide the output of mamba list

BenjaminNavarro, 2022-06-24:
It solves the missing ODE lib error. The other errors are still there:
```
> ign gazebo
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin22311]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin36993]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin39984]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin36827]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin36459]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin91369]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin35473]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin57650]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin42580]
[GUI] [Err] [RenderEngineManager.cc:461] Failed to load plugin [ogre2] : couldn't find shared library.
[GUI] [Err] [RenderUtil.cc:1678] Engine [ogre2] is not supported. Loading OGRE2 instead.
[GUI] [Err] [RenderEngineManager.cc:461] Failed to load plugin [ogre2] : couldn't find shared library.
Stack trace (most recent call last) in thread 280919:
#14   Object "", at 0xffffffffffffffff, in 
#13   Source "../sysdeps/unix/sysv/linux/x86_64/clone.S", line 100, in __clone [0x7f1617f11873]
#12   Source "/usr/src/debug/glibc/nptl/pthread_create.c", line 442, in start_thread [0x7f1617e8c54c]
#11   Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f1612ca190e, in 
#10   Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f1612ca094c, in QThread::exec()
#9    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f1612da899a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#8    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f1612dd8bc0, in QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#7    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f1612dac1bf, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#6    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f1612dabf87, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#5    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Widgets.so.5.12.9", at 0x7f161277a34c, in QApplication::notify(QObject*, QEvent*)
#4    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Widgets.so.5.12.9", at 0x7f1612774821, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#3    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f1612dc0f15, in QObject::event(QEvent*)
#2    Object "/home/benjamin/.conda/envs/ros2_env/lib/ign-gazebo-5/plugins/gui/libGzScene3D.so", at 0x7f15e06cb150, in ignition::gazebo::v5::RenderThread::RenderNext(ignition::gazebo::v5::RenderSync*)
#1    Object "/home/benjamin/.conda/envs/ros2_env/lib/ign-gazebo-5/plugins/gui/libGzScene3D.so", at 0x7f15e06c786f, in ignition::gazebo::v5::IgnRenderer::Initialize()
#0    Object "/home/benjamin/.conda/envs/ros2_env/lib/libignition-gazebo5-rendering.so.5.1.0", at 0x7f15e05f5aff, in ignition::gazebo::v5::RenderUtil::Init()
Segmentation fault (Address not mapped to object [(nil)])
```

Here is the output of `mamba list`:
```
# packages in environment at /home/benjamin/.conda/envs/ros2_env:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
alsa-lib                  1.2.6.1              h7f98852_0    conda-forge
aom                       3.3.0                h27087fc_1    conda-forge
argcomplete               2.0.0              pyhd8ed1ab_0    conda-forge
assimp                    5.0.1                hedfc422_7    conda-forge
attr                      2.5.1                h166bdaf_0    conda-forge
attrs                     21.4.0             pyhd8ed1ab_0    conda-forge
boost-cpp                 1.74.0               h6cacc03_7    conda-forge
bullet-cpp                3.17                 h43a58ef_1    conda-forge
bzip2                     1.0.8                h7f98852_4    conda-forge
c-ares                    1.18.1               h7f98852_0    conda-forge
ca-certificates           2022.6.15            ha878542_0    conda-forge
catkin_pkg                0.5.2              pyhd8ed1ab_0    conda-forge
cli11                     2.2.0                h27087fc_0    conda-forge
cmake                     3.23.2               h5432695_0    conda-forge
console_bridge            1.0.2                h924138e_1    conda-forge
cppzmq                    4.8.1                hf7cf922_0    conda-forge
dartsim                   6.10.1               he39ca3a_1    conda-forge
dbus                      1.13.6               h5008d03_3    conda-forge
docutils                  0.18.1           py38h578d9bd_1    conda-forge
eigen                     3.4.0                h4bd325d_0    conda-forge
empy                      3.3.4              pyh9f0ad1d_1    conda-forge
expat                     2.4.8                h27087fc_0    conda-forge
fcl                       0.6.1                h2cbc392_3    conda-forge
ffmpeg                    4.4.1                hd7ab26d_2    conda-forge
flann                     1.9.1             hfe772e8_1010    conda-forge
font-ttf-dejavu-sans-mono 2.37                 hab24e00_0    conda-forge
font-ttf-inconsolata      3.000                h77eed37_0    conda-forge
[... cut ...]
yaml                      0.2.5                h7f98852_2    conda-forge
yaml-cpp                  0.6.3                he1b5a44_4    conda-forge
zeromq                    4.3.4                h9c3ff4c_1    conda-forge
zipp                      3.8.0              pyhd8ed1ab_0    conda-forge
zlib                      1.2.12               h166bdaf_1    conda-forge
zstd                      1.5.2                h8a70e8d_1    conda-forge
zziplib                   0.13.69              h27826a3_1    conda-forge
```

traversaro, 2022-06-24:
I think this is related to https://github.com/conda-forge/libignition-gazebo-feedstock/issues/27 .

BenjaminNavarro, 2022-06-24:
Doesn't change anything, still complains about not finding ogre2.

But if I edit `~/.ignition/gazebo/gui.config` to use ogre instead of ogre2, as mentioned in the issue you linked, it tries to load ogre, but fails because of a missing shared lib, then tries to load ogre2, still fails and crashes:
```
> ign gazebo                     
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin39500]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin96589]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin87875]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin60670]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin50479]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin61135]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin729]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin59118]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin34163]
Error while loading the library [/home/benjamin/.conda/envs/ros2_env/lib/ign-rendering-5/engine-plugins/libignition-rendering-ogre.so]: libOgreMain.so.1.12.12: cannot open shared object file: No such file or directory
[GUI] [Err] [RenderEngineManager.cc:470] Failed to load plugin [ignition-rendering-ogre] : couldn't load library on path [/home/benjamin/.conda/envs/ros2_env/lib/ign-rendering-5/engine-plugins/libignition-rendering-ogre.so].
[GUI] [Err] [RenderUtil.cc:1678] Engine [ogre] is not supported. Loading OGRE2 instead.
[GUI] [Err] [RenderEngineManager.cc:461] Failed to load plugin [ogre2] : couldn't find shared library.
Stack trace (most recent call last) in thread 282983:
#14   Object "", at 0xffffffffffffffff, in 
#13   Source "../sysdeps/unix/sysv/linux/x86_64/clone.S", line 100, in __clone [0x7f7c50711873]
#12   Source "/usr/src/debug/glibc/nptl/pthread_create.c", line 442, in start_thread [0x7f7c5068c54c]
#11   Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f7c4b4a190e, in 
#10   Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f7c4b4a094c, in QThread::exec()
#9    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f7c4b5a899a, in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>)
#8    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f7c4b5d8bc0, in QEventDispatcherUNIX::processEvents(QFlags<QEventLoop::ProcessEventsFlag>)
#7    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f7c4b5ac1bf, in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*)
#6    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f7c4b5abf87, in QCoreApplication::notifyInternal2(QObject*, QEvent*)
#5    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Widgets.so.5.12.9", at 0x7f7c4af7a34c, in QApplication::notify(QObject*, QEvent*)
#4    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Widgets.so.5.12.9", at 0x7f7c4af74821, in QApplicationPrivate::notify_helper(QObject*, QEvent*)
#3    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7f7c4b5c0f15, in QObject::event(QEvent*)
#2    Object "/home/benjamin/.conda/envs/ros2_env/lib/ign-gazebo-5/plugins/gui/libGzScene3D.so", at 0x7f7c1c6a8150, in ignition::gazebo::v5::RenderThread::RenderNext(ignition::gazebo::v5::RenderSync*)
#1    Object "/home/benjamin/.conda/envs/ros2_env/lib/ign-gazebo-5/plugins/gui/libGzScene3D.so", at 0x7f7c1c6a486f, in ignition::gazebo::v5::IgnRenderer::Initialize()
#0    Object "/home/benjamin/.conda/envs/ros2_env/lib/libignition-gazebo5-rendering.so.5.1.0", at 0x7f7c1c5d2aff, in ignition::gazebo::v5::RenderUtil::Init()
Segmentation fault (Address not mapped to object [(nil)])
```

BenjaminNavarro, 2022-06-24:
After installing ogre 1.12.12 (with `mamba install ogre=1.12.12`) it still crashes but for a different reason:
```
> ign gazebo                            
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin53839]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin12129]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin34365]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin79703]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin95062]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin24654]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin22417]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin33059]
[GUI] [Err] [GuiRunner.cc:158] Failed to get plugin [plugin90755]
terminate called after throwing an instance of 'Ogre::RuntimeAssertionException'
  what():  RuntimeAssertionException: Ogre/ShadowExtrudePointLight not found. Verify that you referenced the 'ShadowVolume' folder in your resources.cfg in initialise at /home/conda/feedstock_root/build_artifacts/ogre_1624011825937/work/OgreMain/src/OgreShadowVolumeExtrudeProgram.cpp (line 70)
Stack trace (most recent call last) in thread 283809:
#10   Object "", at 0xffffffffffffffff, in 
#9    Source "../sysdeps/unix/sysv/linux/x86_64/clone.S", line 100, in __clone [0x7fc762f11873]
#8    Source "/usr/src/debug/glibc/nptl/pthread_create.c", line 442, in start_thread [0x7fc762e8c54c]
#7    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7fc75daa1967, in 
#6    Object "/home/benjamin/.conda/envs/ros2_env/lib/libQt5Core.so.5.12.9", at 0x7fc75da8ec24, in qTerminate()
#5    Source "../../../../libstdc++-v3/libsupc++/eh_terminate.cc", line 58, in terminate [0x7fc75f276575]
#4    Source "../../../../libstdc++-v3/libsupc++/eh_terminate.cc", line 48, in __terminate [0x7fc75f276523]
#3    Source "../../../../libstdc++-v3/libsupc++/vterminate.cc", line 95, in __verbose_terminate_handler [0x7fc75f278035]
#2    Source "/usr/src/debug/glibc/stdlib/abort.c", line 79, in abort [0x7fc762e28534]
#1    Source "../sysdeps/posix/raise.c", line 26, in raise [0x7fc762e3e837]
#0    Source "/usr/src/debug/glibc/nptl/pthread_kill.c", line 44, in __pthread_kill_implementation [0x7fc762e8e36c]
Aborted (Signal sent by tkill() 283740 1000)
```

BenjaminNavarro, 2022-06-24:
The installation fails with:
```
Encountered problems while solving:
  - package ros2-distro-mutex-0.1.0-galactic has constraint ogre 1.12.* conflicting with ogre-1.10.12-hc56bac0_0
```

traversaro, 2022-06-24:
Then I am afraid that at the moment it is not possible to use Ignition Gazebo and ROS2 in the same environment, see https://github.com/conda-forge/libignition-gazebo-feedstock/issues/27 for more details.

traversaro, 2022-06-25:
As you noticed the problem is the missing ogre2/ogre-next . Hopefully this is solved once https://github.com/conda-forge/staged-recipes/pull/19384 is merged and we rebuilt Modern Gazebo (formerly known as Ignition Gazebo) with ogre2/ogre-next support.

BenjaminNavarro, 2022-07-01:
Great! Yes I can definitely switch to humble as I'm not particularly tied to galactic.

I guess the issue can be closed then ?

traversaro, 2022-07-01:
> I guess the issue can be closed then ?

If you are ok with this, yes!