RoboStack/ros-galactic #105: how would i install ros-galactic-octomap-msgs using mamba

an99990, 2022-07-21:
Hey,
so im building stuff inside a docker and uses mamba to install ros stuff. I need octomap-msg, i found that i would need to install ros-galactic-octomap-msgs and ros-galactic-octomap. But mamba doesnt seem to find them ? 

```
RUN conda install mamba -c conda-forge

SHELL ["conda", "run", "-n", "base", "/bin/bash", "-c"]

RUN conda  config --env --add channels conda-forge
# and the robostack channels
RUN conda  config --env --add channels robostack
RUN conda  config --env --add channels robostack-experimental

# Install the version of ROS 
RUN mamba install ros-galactic-desktop 

RUN mamba install compilers cmake pkg-config make ninja colcon-common-extensions ros-galactic-rosbridge-suite ros-galactic-rmw-fastrtps-cpp ros-galactic-octomap-msgs ros-galactic-octomap

RUN conda install dynaconf mesalib ros-galactic-navigation2 octomap
```

Anyone has any idea how i could install it ?

Tobias-Fischer, 2022-07-21:
Hiya,

ros-galactic-octomap does exist (https://anaconda.org/robostack-experimental/ros-galactic-octomap). Could you please provide an error message?

Thanks!

an99990, 2022-07-22:
i found it, sorry

closed 2022-07-22 by an99990