RoboStack/ros-galactic #106: ERROR conda.cli.main_run:execute | inside docker

an99990, 2022-07-22:
Hi
I am having issues building rtabmap and rtabmap_ros inside the conda env.
but removing those packages , everything builds fine ?

```

ARG VARIANT="21.10"
FROM [nvcr.io/nvidia/pytorch:${VARIANT}-py3](http://nvcr.io/nvidia/pytorch:$%7BVARIANT%7D-py3) AS builder

# create conda env to install ros
RUN conda install mamba -c conda-forge
SHELL ["conda", "run", "-n", "base", "/bin/bash", "-c"]
RUN conda  config --env --add channels conda-forge
# and the robostack channels
RUN conda  config --env --add channels robostack
RUN conda  config --env --add channels robostack-experimental
# Install the version of ROS
RUN mamba install ros-galactic-desktop
RUN mamba install compilers cmake pkg-config make ninja colcon-common-extensions ros-galactic-rosbridge-suite ros-galactic-rmw-fastrtps-cpp ros-galactic-octomap-msgs ros-galactic-octomap
RUN conda install dynaconf mesalib mesa-libgl-devel-cos6-x86_64 pyopengl ros-galactic-navigation2
WORKDIR /ros_ws/
RUN bash -c "source /opt/ros/galactic/install/setup.bash; \
  colcon build --packages-up-to \
  zed_wrapper \
  apriltag_msgs \
  rtabmap\
  rtabmap_ros \
```

```
#22 607.8 ERROR conda.cli.main_run:execute(41): `conda run /bin/bash -c bash -c "source /opt/ros/galactic/install/setup.bash;   colcon build --packages-up-to   rtabmap  rtabmap_ros"` failed. (See above for error)
#22 607.8 bash: /opt/ros/galactic/install/setup.bash: No such file or directory
```

Tobias-Fischer, 2022-07-22:
The error is pretty clear: bash: /opt/ros/galactic/install/setup.bash: No such file or directory

There is no need to source the setup.bash in case of RoboStack, simply activate the environment. Please follow our readme where we create a new environment, your code snippet seems to install stuff in the base environment.

Tobias-Fischer, 2022-07-25:
Closing due to inactivity - feel free to reopen if issues persist.

closed 2022-07-25 by Tobias-Fischer