RoboStack/ros-galactic #109: ROS default msgs missing functions

b1n-ch1kn, 2022-10-03:
After installing a new Robostack env for Galactic, building custom msgs that rely on ROS default msgs (like `std_msgs` and `geometry_msgs`) are unable to build correctly.

### Errors

```
(ros_galactic) wsl@Al-Desktop:~/QUTMS_Driverless$ colcon build --packages-select driverless_msgs
Starting >>> driverless_msgs
--- stderr: driverless_msgs
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/cone__functions.c: In function 'driverless_msgs__msg__Cone__are_equal':
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/cone__functions.c:49:8: warning: implicit declaration of function 'geometry_msgs__msg__Point__are_equal'; did you mean 'geometry_msgs__msg__Point__create'? [-Wimplicit-function-declaration]
   49 |   if (!geometry_msgs__msg__Point__are_equal(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        geometry_msgs__msg__Point__create
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/cone__functions.c: In function 'driverless_msgs__msg__Cone__copy':
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/cone__functions.c:70:8: warning: implicit declaration of function 'geometry_msgs__msg__Point__copy'; did you mean 'geometry_msgs__msg__Point__fini'? [-Wimplicit-function-declaration]
   70 |   if (!geometry_msgs__msg__Point__copy(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        geometry_msgs__msg__Point__fini
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/cone_detection_stamped__functions.c: In function 'driverless_msgs__msg__ConeDetectionStamped__are_equal':
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/cone_detection_stamped__functions.c:65:8: warning: implicit declaration of function 'std_msgs__msg__Header__are_equal'; did you mean 'std_msgs__msg__Header__create'? [-Wimplicit-function-declaration]
   65 |   if (!std_msgs__msg__Header__are_equal(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        std_msgs__msg__Header__create
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/cone_detection_stamped__functions.c: In function 'driverless_msgs__msg__ConeDetectionStamped__copy':
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/cone_detection_stamped__functions.c:94:8: warning: implicit declaration of function 'std_msgs__msg__Header__copy'; did you mean 'std_msgs__msg__Header__fini'? [-Wimplicit-function-declaration]
   94 |   if (!std_msgs__msg__Header__copy(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        std_msgs__msg__Header__fini
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/path_stamped__functions.c: In function 'driverless_msgs__msg__PathStamped__are_equal':
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/path_stamped__functions.c:56:8: warning: implicit declaration of function 'std_msgs__msg__Header__are_equal'; did you mean 'std_msgs__msg__Header__create'? [-Wimplicit-function-declaration]
   56 |   if (!std_msgs__msg__Header__are_equal(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        std_msgs__msg__Header__create
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/path_stamped__functions.c: In function 'driverless_msgs__msg__PathStamped__copy':
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/path_stamped__functions.c:79:8: warning: implicit declaration of function 'std_msgs__msg__Header__copy'; did you mean 'std_msgs__msg__Header__fini'? [-Wimplicit-function-declaration]
   79 |   if (!std_msgs__msg__Header__copy(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        std_msgs__msg__Header__fini
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/path_point__functions.c: In function 'driverless_msgs__msg__PathPoint__are_equal':
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/path_point__functions.c:49:8: warning: implicit declaration of function 'geometry_msgs__msg__Point__are_equal'; did you mean 'geometry_msgs__msg__Point__create'? [-Wimplicit-function-declaration]
   49 |   if (!geometry_msgs__msg__Point__are_equal(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        geometry_msgs__msg__Point__create
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/path_point__functions.c: In function 'driverless_msgs__msg__PathPoint__copy':
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/path_point__functions.c:70:8: warning: implicit declaration of function 'geometry_msgs__msg__Point__copy'; did you mean 'geometry_msgs__msg__Point__fini'? [-Wimplicit-function-declaration]
   70 |   if (!geometry_msgs__msg__Point__copy(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        geometry_msgs__msg__Point__fini
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/track_detection_stamped__functions.c: In function 'driverless_msgs__msg__TrackDetectionStamped__are_equal':
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/track_detection_stamped__functions.c:56:8: warning: implicit declaration of function 'std_msgs__msg__Header__are_equal'; did you mean 'std_msgs__msg__Header__create'? [-Wimplicit-function-declaration]
   56 |   if (!std_msgs__msg__Header__are_equal(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        std_msgs__msg__Header__create
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/track_detection_stamped__functions.c: In function 'driverless_msgs__msg__TrackDetectionStamped__copy':
/home/wsl/QUTMS_Driverless/build/driverless_msgs/rosidl_generator_c/driverless_msgs/msg/detail/track_detection_stamped__functions.c:79:8: warning: implicit declaration of function 'std_msgs__msg__Header__copy'; did you mean 'std_msgs__msg__Header__fini'? [-Wimplicit-function-declaration]
   79 |   if (!std_msgs__msg__Header__copy(
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |        std_msgs__msg__Header__fini
---
Finished <<< driverless_msgs [9.23s]

Summary: 1 package finished [9.45s]
  1 package had stderr output: driverless_msgs
```

It finishes building but then will not run.

### Source file

I looked into the source files for the default msgs and the methods that are referenced in the build errors, like `std_msgs__msg__Header__copy`, are missing from the included files.
[... 290 lines cut ...]
/**
 * This functions performs a deep copy, as opposed to the shallow copy that
 * plain assignment yields.
 *
 * \param[in] input The source array pointer.
 * \param[out] output The target array pointer, which must
 *   have been initialized before calling this function.
 * \return true if successful, or false if either pointer
 *   is null or memory allocation fails.
 */
ROSIDL_GENERATOR_C_PUBLIC_std_msgs
bool
std_msgs__msg__Header__Sequence__copy(
  const std_msgs__msg__Header__Sequence * input,
  std_msgs__msg__Header__Sequence * output);

#ifdef __cplusplus
}
#endif

#endif  // STD_MSGS__MSG__DETAIL__HEADER__FUNCTIONS_H_
```

I looked through commit logs and couldn't work out what would have broken it between that timespan. This has occured on all platforms since I began test installing new Galactic environments about a month ago. It does not occur with new Humble environments however, so for any new environments we've installed Humble running on Ubuntu 20.04 in the interum (no compatibility issues so far). 
Let me know if you need more info :)

Tobias-Fischer, 2022-10-03:
Can you try `mamba install ros-galactic-rosidl-generator-c=2.2.1`?

b1n-ch1kn, 2022-10-03:
Amazing, that worked. Thanks!

Tobias-Fischer, 2022-10-03:
Ok. Seems like we need to rebuild the Galactic packages in this case.

@wolfv  what do you think about removing the functionality that automatically builds the latest versions of packages? I think it works to our disadvantage as seen in this issue.

Tobias-Fischer, 2022-10-20:
I deleted the version 2.2.2 of this package, and we don't generate new versions anymore automatically, so this shouldn't be an issue anymore.

closed 2022-10-20 by Tobias-Fischer