RoboStack/ros-galactic #110: install cartographer-ros

husnoo, 2022-12-29:
Hi,

I installed robostack/ros2/galactic using the instructions:
```
source /home/nawal/storage/ros/miniconda/etc/profile.d/conda.sh

conda install mamba -c conda-forge

mamba create -n robostackenv ros-noetic-desktop python=3.9 -c robostack -c robostack-experimental -c conda-forge --no-channel-priority --override-channels
conda activate robostackenv
mamba install rosdep
rosdep init  # note: do not use sudo!
rosdep update

mamba install cmake pkg-config make ninja colcon-common-extensions
```

then tried to install cartographer-ros based on https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html:
```
sudo apt-get install python3-wstool

mkdir catkin_ws
cd catkin_ws
wstool init src
wstool merge -t src https://raw.githubusercontent.com/cartographer-project/cartographer_ros/master/cartographer_ros.rosinstall
wstool update -t src

rosdep init
rosdep update

nano src/cartographer_ros/cartographer_ros.rosinstall
cartographer -> 1.0.9001
cartographer_ros -> 1.0.9003
```
(these are the versions used in an online course)

```
rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
cartographer_ros: Cannot locate rosdep definition for [roslib]
cartographer_ros_msgs: Cannot locate rosdep definition for [message_generation]
cartographer: No definition of [libabsl-dev] for OS [conda]
cartographer_rviz: Cannot locate rosdep definition for [rviz]
```

Thanks for any pointers towards figuring this out!

Tobias-Fischer, 2022-12-29:
Hi, it seems like you installed ros-noetic? This is the ros-galactic repository. Could you please clarify? Thanks :)

husnoo, 2022-12-29:
No, sorry I copied the wrong commands from my notes - I've updated the post now, the conda env is robostackenv2, with ros-galactic-desktop installed.

```
(robostackenv2) nawal@devastator:~ $ rosversion -d
galactic
```

Tobias-Fischer, 2022-12-29:
Any chance to update to humble? You should be able to mamba install ros-humble-cartographer-ros

husnoo, 2022-12-30:
Tried humble:

```(humble) nawal@devastator:~/catkin_ws $ mamba install ros-humble-cartographer-ros -c robostack-humble 

                  __    __    __    __
                 /  \  /  \  /  \  /  \
                /    \/    \/    \/    \
███████████████/  /██/  /██/  /██/  /████████████████████████
              /  / \   / \   / \   / \  \____
             /  /   \_/   \_/   \_/   \    o \__,
            / _/                       \_____/  `
            |/
        ███╗   ███╗ █████╗ ███╗   ███╗██████╗  █████╗
        ████╗ ████║██╔══██╗████╗ ████║██╔══██╗██╔══██╗
        ██╔████╔██║███████║██╔████╔██║██████╔╝███████║
        ██║╚██╔╝██║██╔══██║██║╚██╔╝██║██╔══██╗██╔══██║
        ██║ ╚═╝ ██║██║  ██║██║ ╚═╝ ██║██████╔╝██║  ██║
        ╚═╝     ╚═╝╚═╝  ╚═╝╚═╝     ╚═╝╚═════╝ ╚═╝  ╚═╝

        mamba (1.1.0) supported by @QuantStack

        GitHub:  https://github.com/mamba-org/mamba
        Twitter: https://twitter.com/QuantStack

█████████████████████████████████████████████████████████████

Looking for: ['ros-humble-cartographer-ros']

conda-forge/linux-aarch64                                   Using cache
conda-forge/noarch                                          Using cache
robostack-humble/linux-aarch64                                No change
robostack-humble/noarch                                       No change

Pinned packages:
  - python 3.9.*

Could not solve for environment specs
Encountered problems while solving:
  - nothing provides requested ros-humble-cartographer-ros

The environment can't be solved, aborting the operation

```

husnoo, 2022-12-30:
And from source, same problem:
```
(humble) nawal@devastator:~/catkin_ws $ rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
cartographer: No definition of [libabsl-dev] for OS [conda]
cartographer_ros: Cannot locate rosdep definition for [roslib]
cartographer_ros_msgs: Cannot locate rosdep definition for [catkin]
cartographer_rviz: Cannot locate rosdep definition for [rviz]
Continuing to install resolvable dependencies...

```

husnoo, 2022-12-30:
Sorry should have said, it's raspbian:

```
(humble) nawal@devastator:~/catkin_ws $ uname -a
Linux devastator 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux

(humble) nawal@devastator:~/catkin_ws $ cat /etc/debian_version 
11.5

(humble) nawal@devastator:~/catkin_ws $ dmesg | grep -i rasp
[    0.000000] Machine model: Raspberry Pi 3 Model B Rev 1.2

```