RoboStack/ros-galactic #112: ItemIdentityException when trying to run RViz2

dhrvrc, 2023-10-16:
I followed the instructions for Installation, except used _Micromamba instead of Mamba_
When trying to run RViz2 with the following command (in the **ros_galactic** environment)
`rviz2`

I get the following: 

> terminating due to uncaught exception of type Ogre::ItemIdentityException: ItemIdentityException: Material with the name BaseWhiteNoLighting already exists.

And there's a crash report:
>  rviz2 quit unexpectedly.

**Environment Details:
OS: MacOS 14.0 Sonoma
2022 Apple M2 MacBook Air**

Tobias-Fischer, 2023-10-16:
Hi, we do not support Galactic anymore. Could you please try with Humble? Best, TObi

dhrvrc, 2023-10-17:
> Hi, we do not support Galactic anymore. Could you please try with Humble? Best, TObi

Hey,

The thing I really need it to be ROS Galactic for the purposes of integration with my team. Is there no workaround this any way?

Thanks

dhrvrc, 2023-10-17:
Hey,

The thing I really need it to be Galactic for the purposes of integration with my team. Is there no workaround this any way?

Thanks

Tobias-Fischer, 2023-10-17:
Unfortunately, there is no easy workaround. It is hard to say where the underlying issue is, and given that the problem does not occur in Humble + limited time to look into issues, I am afraid that realistically none of the core maintainers will look into this. You could try different version combinations of ros-galactic-humble and ogre, but it would probably be faster to convince your team to upgrade to Humble which should be straightforward.

dhrvrc, 2023-10-17:
Ah Alright

Thanks for your response :)

closed 2023-10-17 by dhrvrc