RoboStack/ros-galactic #113: Fail to install `rosidl` due to python version incompatible

jc-bao, 2024-01-20:
**Environment Details:** 
- ROS2
- Linux64

**Issue Description:**
I am currently trying to install `ros-galactic-rmw-cyclonedds-cpp` and `ros-galactic-rosidl-generator-dds-idl` into my ROS2 environment. However, I encounter an error indicating an incompatibility with my Python version.

The relevant installation command:

```shell
mamba install ros-galactic-rmw-cyclonedds-cpp ros-galactic-rosidl-generator-dds-idl -f
```

This results in the following error:

```python
Could not solve for environment specs
The following packages are incompatible
└─ ros-galactic-rosidl-generator-dds-idl   is installable with the potential options
   ├─ ros-galactic-rosidl-generator-dds-idl 0.8.0 would require
   │  └─ python_abi 3.8.*  with the potential options
   │     ├─ python_abi 3.8 would require
   │     │  └─ python 3.8.* , which can be installed;
   │     └─ python_abi 3.8, which can be installed;
   └─ ros-galactic-rosidl-generator-dds-idl 0.8.0 would require
      └─ ros2-distro-mutex 0.1.* , which can be installed.
```

From the documentation, it is mentioned that `rosidl` is supported on Linux64. I am unsure why I am encountering the issue. Has anyone else faced a similar problem or might know the solution? Any assistance offered will be greatly appreciated.

[image]

traversaro, 2024-01-20:
What probably happened is that the package was built with Python 3.8, and never built again after the Python 3.9 migration. You may try to create a fresh environment specifying during the environment creation that you want `ros-galactic-rosidl-generator-dds-idl`, so that the solver is aware of this and installs the correct Python version. However, as far as I remember Python 3.8 builds had problems, so you may encounter problem doing that. For sure the best option would be to fix https://github.com/RoboStack/ros-humble/issues/125, but we would probably need to fix https://github.com/RoboStack/ros-humble/pull/124 before, that may not be trivial to solve.

jc-bao, 2024-01-20:
I see. Thank you very much for your prompt response. I've successfully installed the package using Python 3.8, but it appears that rviz isn't functioning properly.

In addition, would you mind providing more details about the challenge mentioned in issue#124? Does this imply that we can't install the package even if we opt to build it from the source in `humble`?

traversaro, 2024-01-20:
> In addition, would you mind providing more details about the challenge mentioned in issue#124? Does this imply that we can't install the package even if we opt to build it from the source in `humble`?

I think if you build it from source in humble everything works fine. The https://github.com/RoboStack/ros-humble/pull/124 tracks an attempt of fixing the CI for building again all packages in `humble`. If that works, then we can add `rosidl-generator-dds-idl` (or perhaps we can directly add that in the existing master in humble, @Tobias-Fischer probably knows more).

jc-bao, 2024-01-20:
Got it. Thank you for the information!

closed 2024-01-20 by jc-bao